NXPNFCLinux/nxp-pn5xx #1: What if there are multiple devices on the same i2c bus?

yaoz39, 2015-09-24:
I am in the process of evaluating PN7120 as an NFC reader.  Based on AN11697.pdf, the PN5xx I2C driver should be used.  But I have other I2C devices on the same I2C bus.  If PN5xx I2C driver is loaded, do I still have access to the other I2C devices the same as before?

Thanks in advance.

jgeslin, 2015-09-25:
The I2C bus can be shared with others slave devices while address is different. Only the GPIO connections (IRQ and VEN) must be exclusive. 
The PN7120 IC has configureable I2C address (0x28 or 0x29) though one pin positioning (tight to ground or to Pvdd reference) allowing to change it in case of conflict with other device connected on the same bus.

yaoz39, 2015-09-25:
Thanks for the quick reply.

Currently, our device tree has entries like
                i2c: i2c@29 {
                        clock-frequency = <50000>;
                        compatible = "cdns,i2c-r1p10";
                        reg = <0x29>;
                } ;

If I add the sample pn547 device tree
        pn547: pn547@29 {
		compatible = "nxp,pn547";
		reg = <0x29>;
		clock-frequency = <400000>;
		interrupt-gpios = <&gpio2 17 0>;
		enable-gpios = <&gpio4 21 0>;
	};

Will it cause any conflict?  Can both i2c device drivers loaded at the same time?  Can I still access other devices as before through /dev/i2c?

Sorry for more questions.

jgeslin, 2015-09-28:
You cannot definitely use 2 slave devices on the same I2C bus with the same I2C address, unless you can enable only one at a time.
I would recommend you to use the other possible I2C address of PN7120 (0x28), replacing "29" by "28" in the device tree sample and fixing I2CADR0 pin (B2) to 0 (GND). See http://www.nxp.com/documents/application_note/AN11565.pdf for more details.
In case you are using OM5577/PN7120S demo kit (see http://www.nxp.com/board/OM5577.html), the address is already fixed to 0x28.

closed 2015-10-12 by jgeslin