NXPNFCLinux/nxp-pn5xx #2: Bug on regulator settings

diegosueiro, 2015-10-09:
Here is the patch:

Index: git/pn5xx_i2c.c
===================================================================
--- git.orig/pn5xx_i2c.c    2015-10-09 14:18:44.796257976 -0300
+++ git/pn5xx_i2c.c 2015-10-09 14:19:47.960255599 -0300
@@ -609,7 +609,7 @@
    pn54x_dev->clkreq_gpio = pdata->clkreq_gpio;
    pn54x_dev->pvdd_reg = pdata->pvdd_reg;
    pn54x_dev->vbat_reg = pdata->vbat_reg;
-   pn54x_dev->pmuvcc_reg = pdata->vbat_reg;
+   pn54x_dev->pmuvcc_reg = pdata->pmuvcc_reg;
    pn54x_dev->sevdd_reg = pdata->sevdd_reg;

    pn54x_dev->client = client;

jgeslin, 2015-10-12:
Patch integrated in the mainline.
Thanks for your help !

closed 2015-10-12 by jgeslin

diegosueiro, 2015-10-12:
You are welcome.