NXPNFCLinux/nxp-pn5xx #3: Software stack from NXP or Intel?

diegosueiro, 2015-10-13:
I saw that there is some effort driven by Intel for bringing "native" support of NFC on the Linux Kernel and userspace services and tools (Linux* NFC):
https://01.org/linux-nfc
https://git.kernel.org/cgit/linux/kernel/git/sameo/nfc-next.git/tree/drivers/nfc

Basically, I'm developing an application with PN7120 using Read/Write and P2P modes.
I'd like to know which is the best path to follow.

jgeslin, 2015-10-13:
Also the PN7120 is supported by the Linux* NFC stack (through nxp-nci lower interface), NXP recommends use of the Linux libnfc-nci for PN7120 integration under Linux system.
See all related documents (especially AN11697) on NXP web site: http://www.nxp.com/products/interface_and_connectivity/nfc_and_reader_ics/nfc_controller_solutions/PN7120A0EV.html

diegosueiro, 2015-10-13:
@jgeslin , thanks for the info.

According to the material bellow, the NXP NCI Extension (pag 28) implements additional proprietary features, and I believe that it is developed both on nxp-pn5xx (kernel driver) and linux_libnfc-nci.
http://www.themobileknowledge.com/sites/default/files/PN7120-%20Best%20plugn%20play%20full%20NFC%20solution.pdf

Do you know if these features are implemented on Linux* NFC too?
The use of neard seems to be a better approach since it uses a D-Bus API.

Also on the document bellow (pag 30) advises to use Linux* NFC:
http://www.nxp.com/documents/brochure/939775017634.pdf

jgeslin, 2015-10-13:
The NXP NCI Extensions stated in the provided materials relates to PN7120 implementation, not to the Linux stack. 
However, the Linux libnfc-nci stack offers interface to those NXP's proprietary features while the Linux* NFC stack doesn't; i.e. support of Reader/Writer of MIFARE Classic card, Host Card Emulation are not yet granted in Linux* NFC stack.
Furthermore, there are also a few relevant functionalities not offered by the Linux* NFC stack; i.e. capability to retrieve remote card UID and to read/write other than NDEF data.

The advise to use Linux* NFC is prior to NXP's decision to discontinue contribution to it. The current NXP's recommendation is to go with Linux libnfc-nci stack.

diegosueiro, 2015-10-13:
Thanks again.

I'll try to prototype an application in P2P mode.
I believe that nfcDemoApp from libnfc-nci does not implement it. Is it correct?

jgeslin, 2015-10-13:
The nfcDemoApp demonstrates all features supported by PN7120 (R/W, P2P, HCE). It shows how easy it is to use the library API to create NFC related scenario.

closed 2015-10-13 by diegosueiro