NXPNFCLinux/nxp-pn5xx #4: NFC Linux subsystem

ContribOpenSource, 2016-02-07:
Hello,

it seems that this driver is not compliant with kernel NFC subsystem:
   https://www.kernel.org/doc/Documentation/networking/nfc.txt
   https://www.kernel.org/doc/Documentation/nfc/nfc-pn544.txt
   https://01.org/linux-nfc/documentation/hardware-support
Consequently, we can not use "standard" userspace socket calls, nor rfkill, etc.

Is there a plan to use NFC kernel subsystem for PN7120 device driver?

Regards

jgeslin, 2016-02-08:
Hi,

PN7120 is already supported by the Linux NFC kernel sub-system, through the "nxp-nci" NFC driver.
However, this is not the recommended solution by NXP. See [AN11697 - PN7120 Linux Software Stack Integration Guidelines.pdf](https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/doc/AN11697%20-%20PN7120%20Linux%20Software%20Stack%20Integration%20Guidelines.pdf) for more details.

Regards

ContribOpenSource, 2016-02-08:
Hello,

thanks for your quick reply. :)

> However, this is not the recommended solution by NXP.

Can you please list the reasons to not use NFC kernel subsystem?
It seems to be a good choice for "basic" functionality like a tag reader, no?

Regards

jgeslin, 2016-02-09:
Hi,

NXP participated to improvments of the Linux NFC Open source stack (v1.2.0 version).
At this time (mid-2015), the view of NXP was that a user mode stack derived from its available and proven Android stack was the right way forward to support the implementation of a broad range of use cases in Linux environment (mostly motivated by the lack of maturity and missing features of the Open source stack as well as by the non reliable plan for updates from the community).

Regards

jgeslin, 2016-10-18:
Can we consider this item closed ?

closed 2018-09-04 by jgeslin