NXPNFCLinux/nxp-pn5xx #5: Stack not working on kernel 4.1.17+

mihaipop11, 2016-02-18:
pn544 node in /dev fails to be added on newer kernels, tested on 4.1.17+ with raspberry pi b plus
also using the nxp demo image the nfc works but after apt-get upgrade, again the same problem, the pn544 node is not created in /dev. Is there a workaround for this issue?

jgeslin, 2016-02-23:
pn544 node is not part of the linux kernel delivery (neither the RPi derived one). It must be added manually as described in Chapter 3 of related documentation [here](https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/doc/AN11697%20-%20PN7120%20Linux%20Software%20Stack%20Integration%20Guidelines.pdf).
Command "apt-get upgrade" lead to update of the kernel of the board and as such the new installed kernel doesn't include nxp-pn5xx module.

jgeslin, 2016-10-18:
Can we consider this item closed ?

closed 2018-07-22 by mihaipop11