NXPNFCLinux/nxp-pn5xx #7: Kernel panic on Android6.0

wintonliuwen, 2016-06-13:
Hi ,

I used that driver on Android5.1 is OK. 
After using Android6.0 with that driver, the kernel Oops:
Android5.1 kernel version:**3.14.41**
Android6.0 kernel version:**3.14.55**
Does that driver tested on Android 6.0?

[    9.762432] =========could not get default pinstate
[    9.767273] BUG: unable to handle kernel paging request at fffffff9
[    9.773302] IP: [<80adb732>] pinctrl_select_state+0x92/0x190
[    9.778880] *pdpt = 00000000013cd001 *pde = 00000000013ce063 *pte = 0000000000000000 
[    9.786598] Oops: 0000 [#1] PREEMPT SMP 
[    9.790457] Modules linked in: pn5xx_i2c(+) gt9xx kxcjk_1013 gnss_drv(O) vidt_driver(O)
[    9.798353] CPU: 0 PID: 171 Comm: hald Tainted: G        W  O 3.14.55-i386-gbfaf348 #2
[    9.806153] task: b42c7000 ti: b42ec000 task.ti: b42ec000
[    9.811470] EIP: 0060:[<80adb732>] EFLAGS: 00010246 CPU: 0
[    9.816880] EIP is at pinctrl_select_state+0x92/0x190
[    9.821846] EAX: b2c09500 EBX: ffffffed ECX: b2c09500 EDX: ffffffed
[    9.828021] ESI: b42bd300 EDI: fffffff9 EBP: b42edcbc ESP: b42edc9c
[    9.834200]  DS: 007b ES: 007b FS: 00d8 GS: 0033 SS: 0068
[    9.839515] CR0: 80050033 CR2: fffffff9 CR3: 342b2000 CR4: 001027b0
[    9.845765] DR0: 8088a515 DR1: 00000006 DR2: 00000001 DR3: 00000068
[    9.851862] DR6: 00000037 DR7: 342b2000
[    9.855634] Stack:
[    9.857607]  b42edcbc 80f10a8f 00000000 00000000 b2c09500 b42edcec b42bd300 802bac00
[    9.865240]  b42edd18 c5181b75 c5182608 c5182700 00000012 ffffffed b42bd344 802bac1c
[    9.872871]  b2c09500 b778cf28 00000000 00000000 00000000 00000012 ffffffff ffffffff
[    9.880506] Call Trace:
[    9.882924]  [<80f10a8f>] ? printk+0x3d/0x3f
[    9.887129]  [<c5181b75>] pn54x_probe+0x1e5/0x5d9 [pn5xx_i2c]
[    9.892787]  [<c5181990>] ? pn54x_dev_write+0xb0/0xb0 [pn5xx_i2c]
[    9.898788]  [<80c5a215>] i2c_device_probe+0x85/0xd0
[    9.903674]  [<8099e9b5>] ? sysfs_create_link+0x25/0x40
[    9.908821]  [<80b6b0b3>] driver_probe_device+0x93/0x3a0
[    9.914052]  [<80d50d32>] ? of_match_node+0x32/0x40
[    9.918857]  [<80b6b400>] ? __device_attach+0x40/0x40
[    9.923829]  [<80b6b471>] __driver_attach+0x71/0x80
[    9.928630]  [<80b6b400>] ? __device_attach+0x40/0x40
[    9.933604]  [<80b69567>] bus_for_each_dev+0x47/0x80
[    9.938493]  [<80b6ab8e>] driver_attach+0x1e/0x20
[    9.943123]  [<80b6b400>] ? __device_attach+0x40/0x40
[    9.948097]  [<80b6a7e7>] bus_add_driver+0x157/0x230
[    9.952989]  [<80b6ba19>] driver_register+0x59/0xe0
[    9.957789]  [<c507e000>] ? 0xc507dfff
[    9.961475]  [<c507e000>] ? 0xc507dfff
[    9.965166]  [<80c5916a>] i2c_register_driver+0x2a/0xd0
[    9.970314]  [<c507e029>] pn54x_dev_init+0x29/0x2b [pn5xx_i2c]
[    9.976059]  [<80800442>] do_one_initcall+0xa2/0x160
[    9.980947]  [<8086be67>] ? preempt_count_sub+0x67/0xf0
[    9.986095]  [<80862407>] ? notifier_call_chain+0x47/0x90
[    9.991409]  [<80933cf4>] ? kfree+0x84/0x190
[    9.995614]  [<808625f4>] ? __blocking_notifier_call_chain+0x44/0x60
[   10.001875]  [<808a9e39>] load_module+0x19c9/0x2280
[   10.006684]  [<808aa855>] SyS_finit_module+0x75/0xc0
[   10.011564]  [<8091467b>] ? vm_mmap_pgoff+0x7b/0xa0
[   10.016372]  [<80f1e63a>] sysenter_do_call+0x12/0x12
[   10.021247] Code: 39 4a 14 74 d4 8b 12 39 d7 75 ec 89 f0 e8 97 2b 00 00 8b 45 f0 8b 36 8b 40 14 8d 50 0c 39 d6 75 c1 8b 45 f0 c7 40 14 00 00 00 00 <8b> 73 0c 39 f7 75 27 e9 ba 00 00 00 66 90 72 36 83 f8 04 77 31
[   10.039770] EIP: [<80adb732>] pinctrl_select_state+0x92/0x190 SS:ESP 0068:b42edc9c

Winton

jgeslin, 2016-06-13:
The driver is independent of the Android version but depends on the kernel version.
According the minor differences in your kernel version (3.14.41 vs 3.14.55) you should not get any issue wrt the pn5xx_i2c driver integration.
Did you built-it from the 3.14.55 kernel source (including the device-tree insertion) or did you simply reused the .ko and dtb built with 3.14.41 ?

wintonliuwen, 2016-06-13:
I build it in 3.14.55 kernel source and the device-tree is also built.

wintonliuwen, 2016-06-15:
Hi Jgeslin,

I had a strange problem. When I plug off NFC module, there is still  /dev/pn544 device. Is that driver will check whether there is NFC module?

Winton

jgeslin, 2016-06-15:
This is expected behavior. Indeed, if you look to "probe" function of the driver, no communication with the device is done to check presence of the NFC Controller. This will only appears when the application (through the NFC library) will trigger communication.

Regards,
Jeremy.

wintonliuwen, 2016-06-23:
Hi Jgeslin,

If I port NFC in Marshmallow. Which driver should I use?
There is another nxp-pn5xx driver here.
https://github.com/NXPNFCProject/NXPNFC_I2CDriver

I ported libnfc-nci, framework, app for Marshmallow and with your driver:
https://github.com/NXPNFCLinux/nxp-pn5xx
Got that issue: bad ioctl
[53789.646379] pn54x_dev_ioctl bad ioctl 1074063621
[53789.651778] pn54x_dev_read : reading 3 bytes.

jgeslin, 2016-06-23:
Hi,

You can keep kernel module from [https://github.com/NXPNFCLinux/nxp-pn5xx](https://github.com/NXPNFCLinux/nxp-pn5xx). The other one includes modification related to a different IC which integrates a secure element. The Marshmallow stack is also supporting this different IC, so for PN7120 you must disable secure element support from the Android.mk files (external/libnfc-nci and exteranl/libnfc-nci/halimpl/pn54x), replacing "LOCAL_CFLAGS += -DNFC_NXP_ESE=TRUE" by "LOCAL_CFLAGS += -DNFC_NXP_ESE=FALSE".

Jeremy.

wintonliuwen, 2016-06-28:
Hi Jeremy,

I have updated the Android.mk as your guide. Now could read FW Version:
**NxpNci> FW Version: 8.2.2**
But couldn't enable NFC in settings.

I found  after updated libnfc-nci,base-nfc for Marshmallow, the IIC commands are also different from Lollipop. 
That's the nfc command for marshmallow:
_01-01 00:06:06.262 D/NxpNciX ( 1055): len =   4 > 20000100
01-01 00:06:06.289 D/NxpNciR ( 1055): len =   6 > 400003001000
01-01 00:06:06.346 D/NxpNciX ( 1055): len =   3 > 200100
01-01 00:06:06.373 D/NxpNciR ( 1055): len =  26 > 40011700030E03000600010203818202C800FF02000405080202
01-01 00:06:06.377 D/NxpNciX ( 1055): len =   6 > 20030301A0EC
01-01 00:06:06.391 D/NxpNciR ( 1055): len =   9 > 4003060001A0EC0100
01-01 00:06:06.441 D/NxpNciX ( 1055): len =   4 > 20000100
01-01 00:06:06.446 D/NxpNciR ( 1055): len =   6 > 400003001000
01-01 00:06:06.449 D/NxpNciX ( 1055): len =   3 > 200100
01-01 00:06:06.572 D/NxpNciR ( 1055): len =  26 > 40011700030E03000600010203818202C800FF02000405080202
01-01 00:06:06.575 D/NxpNciX ( 1055): len =  10 > 21030703800181018201
01-01 00:06:06.614 D/NxpNciR ( 1055): len =   4 > 41030100
01-01 00:06:06.620 D/NxpNciX ( 1055): len =   4 > 20000100
01-01 00:06:06.660 D/NxpNciR ( 1055): len =   6 > 400003001000_

**And that's the nfc command for Lollipop, which works OK.**
D/NxpNciX ( 1437): len =   4 > 22000101
D/NxpNciR ( 1437): len =   5 > 4200020001
D/NxpNciR ( 1437): len =   8 > 6200050101010100
D/NxpNciX ( 1437): len =   9 > 200406030101020101
D/NxpNciR ( 1437): len =   7 > 40040400FF0103
D/NxpNciX ( 1437): len =   5 > 0300028103
D/NxpNciR ( 1437): len =   6 > 600603010301
D/NxpNciR ( 1437): len =   5 > 0300028180
D/NxpNciX ( 1437): len =   7 > 030004810103C0
D/NxpNciR ( 1437): len =   6 > 600603010301
D/NxpNciR ( 1437): len =   5 > 0300028180
D/NxpNciX ( 1437): len =  14 > 03000B8101012FB5FFFF28B5FFFF
D/NxpNciR ( 1437): len =   6 > 600603010301
D/NxpNciR ( 1437): len =   5 > 0300028180
D/NxpNciX ( 1437): len =   5 > 2201020101
D/NxpNciR ( 1437): len =   4 > 42010100
D/NxpNciX ( 1437): len =   6 > 030003810204
D/NxpNciR ( 1437): len =   6 > 600603010301
D/NxpNciR ( 1437): len =   6 > 030003818000
D/NxpNciX ( 1437): len =   6 > 20030301A0F0
D/NxpNciR ( 1437): len =  76 > 4003490001A0F04400FF000000000000000000000000000000000000000000000000000000000000EE000103030000FF0000000000000000E400000000000000000000000000000000030300
D/NxpNciX ( 1437): len =   8 > 20030502A0ECA0ED
D/NxpNciR ( 1437): len =  13 > 40030A0002A0EC0100A0ED0100
D/NxpNciX ( 1437): len =   6 > 20030301A0EF
D/NxpNciR ( 1437): len =  76 > 4003490001A0EF4400FF000000000000000000000000000000000000000000000000000000000000EE000103030000FF0000000000000000E400000000000000000000000000000000030300
D/NxpNciX ( 1437): len =   8 > 20020501A0C00104
D/NxpNciR ( 1437): len =   5 > 4002020000
D/NxpNciX ( 1437): len =   8 > 20020501A0070103
D/NxpNciR ( 1437): len =   5 > 4002020000
D/NxpNciR ( 1437): len =   4 > 4F150100
D/NxpNciX ( 1437): len =   4 > 2F150100
D/NxpNciX ( 1437): len =  20 > 2101110003010300410401030041A00103000105
D/NxpNciR ( 1437): len =   4 > 41010100
D/NxpNciX ( 1437): len =  19 > 20021005320120300104310100380101500100
D/NxpNciR ( 1437): len =   5 > 4002020000
D/NxpNciX ( 1437): len =   8 > 200205010002F401
D/NxpNciR ( 1437): len =   5 > 4002020000
D/NxpNciX ( 1437): len =   8 > 2103050280018101
D/NxpNciR ( 1437): len =   4 > 41030100
D/NxpNciX ( 1437): len =   4 > 21060100
D/NxpNciR ( 1437): len =   4 > 41060100
D/NxpNciX ( 1437): len =   5 > 20030201B9
D/NxpNciR ( 1437): len =   7 > 4003040901B900
D/NxpNciX ( 1437): len =  26 > 20021701291446666D010111020207FF03020013040164070103
D/NxpNciR ( 1437): len =   5 > 4002020000
D/NxpNciX ( 1437): len =  29 > 20021A02611446666D010111020207FF03020013040164070103600107
D/NxpNciR ( 1437): len =   5 > 4002020000
D/NxpNciX ( 1437): len =  16 > 20020D04320160310100380101500102
D/NxpNciR ( 1437): len =   5 > 4002020000
D/NxpNciX ( 1437): len =  28 > 2103190C000101010201030105018001810182018301850106017701
D/NxpNciR ( 1437): len =   4 > 41030100
D/NxpNciR ( 1437): len =  32 > 61051D01020400FF010C44030704287D8A843C80012000000006057577810280

If the command are different, could the NFC controller be compatible with the Marshmallow commands?

Winton

wintonliuwen, 2016-06-28:
Marshmallow NFC log:
01-01 00:06:05.847 D/NfcService( 1055): NFC is on. Doing normal stuff
01-01 00:06:05.847 I/NfcService( 1055): Enabling NFC
01-01 00:06:05.847 D/BrcmNfcJni( 1055): nfcManager_getChipVer: enter
01-01 00:06:05.847 D/BrcmNfcJni( 1055): 1: nfcManager_getChipVer
01-01 00:06:05.858 D/BrcmNfcJni( 1055): nfcManager_getNfcInitTimeout: enter
01-01 00:06:05.858 D/BrcmNfcJni( 1055): NAME_NXP_DEFAULT_NFCEE_DISC_TIMEOUT not found
01-01 00:06:05.858 D/BrcmNfcJni( 1055): NAME_NXP_DEFAULT_NFCEE_TIMEOUT not found
01-01 00:06:05.858 D/BrcmNfcJni( 1055):  gNfcInitTimeout = 0: gdisc_timeout = 0 nfcManager_getNfcInitTimeout
01-01 00:06:05.859 I/NfcService( 1055): Enabling NFC timeout90000
01-01 00:06:05.864 I/LatinIME:LogUtils( 1021): Dictionary info: dictionary = userunigram.en_US ; version = 946685165 ; date = ?
01-01 00:06:05.872 D/BrcmNfcJni( 1055): nfcManager_doInitialize: enter; ver=NFCDROID-AOSP_M_00.01 nfa=NFA_PI_1.03.66+ NCI_VERSION=0x10
01-01 00:06:05.872 D/BrcmNfcJni( 1055): nfcManager_doInitialize:  MW Version: NFC_NCIHALx_AR3C.4.5.0
01-01 00:06:05.872 D/BrcmNfcJni( 1055): PowerSwitch::initialize: level=PS-FULL (1)
01-01 00:06:05.872 D/NfcAdaptation( 1055): const CNfcParam* CNfcConfig::find(const char*) const found SCREEN_OFF_POWER_STATE=(0x1)
01-01 00:06:05.872 D/BrcmNfcJni( 1055): PowerSwitch::initialize: desired screen-off state=1
01-01 00:06:05.872 D/NfcAdaptation( 1055): NfcAdaptation::Initialize: enter
01-01 00:06:05.872 E/NfcAdaptation( 1055): NfcAdaptation::Initialize: ver=NFCDROID-AOSP_M_00.01 nfa=NFA_PI_1.03.66+
01-01 00:06:05.872 D/NfcAdaptation( 1055): const CNfcParam* CNfcConfig::find(const char*) const found PROTOCOL_TRACE_LEVEL=(0xFFFFFFFF)
01-01 00:06:05.872 D/NfcAdaptation( 1055): const CNfcParam* CNfcConfig::find(const char*) const found NFA_MAX_EE_SUPPORTED=(0x1)
01-01 00:06:05.872 D/NfcAdaptation( 1055): NfcAdaptation::Initialize: Overriding NFA_EE_MAX_EE_SUPPORTED to use 1
01-01 00:06:05.872 D/NfcAdaptation( 1055): const CNfcParam* CNfcConfig::find(const char*) const found APPL_TRACE_LEVEL=(0xFF)
01-01 00:06:05.872 D/BrcmNfcJni( 1055): initializeGlobalAppLogLevel: level=5
01-01 00:06:05.872 D/        ( 1055): verify_stack_non_volatile_store
01-01 00:06:05.872 D/        ( 1055): crcChecksumVerifyIntegrity: filename=/data/nfc/nfaStorage.bin1
01-01 00:06:05.872 D/        ( 1055): crcChecksumVerifyIntegrity: filename=/data/nfc/nfaStorage.bin2
01-01 00:06:05.872 D/        ( 1055): crcChecksumVerifyIntegrity: filename=/data/nfc/nfaStorage.bin3
01-01 00:06:05.872 D/        ( 1055): crcChecksumVerifyIntegrity: filename=/data/nfc/nfaStorage.bin4
01-01 00:06:05.872 D/        ( 1055): crcChecksumVerifyIntegrity: filename=/data/nfc/nfaStorage.bin5
01-01 00:06:05.872 D/        ( 1055): delete_stack_non_volatile_store
01-01 00:06:05.874 D/NfcAdaptation( 1055): NfcAdaptation::Thread: enter
01-01 00:06:05.874 D/NfcAdaptation( 1055): NfcAdaptation::NFCA_TASK: enter
01-01 00:06:05.874 I/BrcmNfcNfa( 1055): GKI_run(): Start/Stop GKI_timer_update_registered!
01-01 00:06:05.874 I/BrcmNfcNfa( 1055): NFC_TASK started.
01-01 00:06:05.874 D/NfcAdaptation( 1055): NfcAdaptation::Thread: exit
01-01 00:06:05.875 D/NfcAdaptation( 1055): NfcAdaptation::InitializeHalDeviceContext: enter
01-01 00:06:05.875 D/NfcAdaptation( 1055): const CNfcParam* CNfcConfig::find(const char*) const found NCI_HAL_MODULE=nfc_nci.pn54x
01-01 00:06:05.877 I/HAL client( 1055): HAL_IPC_MAP failed: No such file or directory
01-01 00:06:05.877 E/HAL     ( 1055): Could not find a HAL module for nfc_nci.pn54x
01-01 00:06:05.898 D/NxpNfcNciHal( 1055): nfc_open: enter; name=nci
01-01 00:06:05.898 D/NxpNfcNciHal( 1055): nfc_open: exit 0
01-01 00:06:05.898 D/NfcAdaptation( 1055): NfcAdaptation::InitializeHalDeviceContext: exit
01-01 00:06:05.898 D/NfcAdaptation( 1055): NfcAdaptation::Initialize: exit
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_dm_init ()
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_sys_register () id=1, enable_cplt_mask=0x0
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_sys_register () id=3, enable_cplt_mask=0x8
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_rw_init ()
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_sys_register () id=6, enable_cplt_mask=0x48
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_ce_init ()
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_sys_register () id=7, enable_cplt_mask=0xc8
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_ee_init ()
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_sys_register () id=2, enable_cplt_mask=0xcc
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_hci_init ()
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): nfa_sys_register () id=8, enable_cplt_mask=0x1cc
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): LLCP - llcp_init ()
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): num_rx_buff = 21, rx_congest_start = 14, rx_congest_end = 10, max_num_ll_rx_buff = 6
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): max_num_tx_buff = 49, max_num_ll_tx_buff = 14
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): LLCP_RegisterServer (): SAP:0x1, link_type:0x2, ServiceName:<urn:nfc:sn:sdp>
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): LLCP_RegisterServer (): Registered SAP = 0x01
01-01 00:06:05.898 I/BrcmNfcNfa( 1055): NFA_Enable ()
01-01 00:06:05.899 D/NfcAdaptation( 1055): const CNfcParam* CNfcConfig::find(const char*) const found APPL_TRACE_LEVEL=(0xFF)
01-01 00:06:05.899 D/BrcmNfcJni( 1055): initializeGlobalAppLogLevel: level=5
01-01 00:06:05.899 I/BrcmNfcNfa( 1055): NFA got event 0x0100
[... cut ...]
01-01 00:06:06.802 D/        ( 1055): int CNxpNfcConfig::updateTimestamp() file /data/nfc/libnfc-nxpConfigState.bin not exist, creat it.
01-01 00:06:06.807 W/NfcService( 1055): Error enabling NFC
01-01 00:06:06.807 D/BrcmNfcJni( 1055): nfcManager_getChipVer: enter
01-01 00:06:06.807 D/NxpExtns( 1055): const CNxpNfcParam* CNxpNfcConfig::find(const char*) const found NXP_NFC_CHIP=(0x1)
01-01 00:06:06.807 D/BrcmNfcJni( 1055): 1: nfcManager_getChipVer
01-01 00:06:06.811 I/BrcmNfcNfa( 1055): GKI TASK_DEAD received. exit thread 4...
01-01 00:06:06.811 D/NfcAdaptation( 1055): NfcAdaptation::NFCA_TASK: exit

jgeslin, 2016-07-05:
Please try out with attached patch: [PN7120_phNxpNciHal_patch.txt](https://github.com/NXPNFCLinux/nxp-pn5xx/files/347464/PN7120_phNxpNciHal_patch.txt)

wintonliuwen, 2016-07-08:
Hi 
[](url)

wintonliuwen, 2016-07-10:
Hi  Jeremy,

I updated some code to make IIC command on M are same as Lollipop.Now I could enable NFC settings.
But reading is abnormal, there are too many interrupts when touching NFC module.So using TagInfo app will be reading all the time. Do you know what's the problem? 
That's the IIC[ signal ](https://github.com/wintonliuwen/nfctest/blob/master/TouchSignal.jpg)I captured.
And the IIC command I updated on M is [that](https://github.com/wintonliuwen/nfctest/blob/master/nfc.M_wrong_command.txt).
That's the right Lollipop IIC [command](https://github.com/wintonliuwen/nfctest/blob/master/nfc.L_sucess_command.txt).

Winton

wintonliuwen, 2016-07-13:
01-01 00:03:03.788 D/NxpTml  ( 1047): PN54X - Write requested.....
01-01 00:03:03.788 D/NxpTml  ( 1047): PN54X - Invoking I2C Write.....
01-01 00:03:03.788 D/NxpTml  ( 1047): PN54X - Read requested.....
01-01 00:03:03.788 D/NxpTml  ( 1047): PN54X - Invoking I2C Read.....
01-01 00:03:03.790 D/NxpNciX ( 1047): len =   3 > 200100
01-01 00:03:03.790 D/NxpTml  ( 1047): PN54X - I2C Write successful.....
01-01 00:03:03.790 D/NxpTml  ( 1047): PN54X - Posting Fresh Write message.....
01-01 00:03:03.790 D/NxpTml  ( 1047): PN54X - Tml Writer Thread Running................
01-01 00:03:03.790 D/NxpHal  ( 1047): write successful status = 0x0
01-01 00:03:03.794 D/NxpTml  ( 1047): PN54X - I2C Read successful.....
01-01 00:03:03.794 D/NxpNciR ( 1047): len =  26 > 40011700030E03000600010203818202C800FF02000405080202
01-01 00:03:03.794 D/NxpTml  ( 1047): PN54X - Posting read message.....
01-01 00:03:03.794 D/NxpHal  ( 1047): read successful status = 0x0
01-01 00:03:03.794 D/NxpHal  ( 1047): NxpNci> FW Version: 8.2.2
01-01 00:03:03.794 D/NfcAdaptation( 1047): NfcAdaptation::HalDeviceContextDataCallback: len=26
01-01 00:03:03.794 I/BrcmNfcNfa( 1047): NFC received rsp gid:0
01-01 00:03:03.794 I/BrcmNfcNfa( 1047): nci_proc_core_rsp opcode:0x1
01-01 00:03:03.794 D/NxpTml  ( 1047): PN54X - Read requested.....
01-01 00:03:03.794 D/NfcAdaptation( 1047): NfcAdaptation::HalIoctl
01-01 00:03:03.794 D/NxpTml  ( 1047): PN54X - Invoking I2C Read.....
01-01 00:03:03.794 D/NxpHal  ( 1047): phNxpNciHal_ioctl : enter - arg = 7
01-01 00:03:03.794 D/NxpFwDnld( 1047): Allocating Mem for Dnld Context..
01-01 00:03:03.798 E/        ( 1029): release-version: 1.12 
01-01 00:03:03.798 E/        ( 1029):  build time: '11:33:39' - 'Feb 15 2016'
01-01 00:03:03.807 E/NxpTml  ( 1047): _i2c_read() [hdr] errno : 79
01-01 00:03:03.807 E/NxpTml  ( 1047): PN54X - Error in I2C Read.....
01-01 00:03:03.807 D/NxpTml  ( 1047): PN54X - Read requested.....
01-01 00:03:03.807 D/NxpFwDnld( 1047): @@@/system/vendor/firmware/libpn547_fw.so
01-01 00:03:03.807 E/HAL client( 1029): No payload in reply
01-01 00:03:03.807 E/NxpFwDnld( 1047): NULL handler : unable to load the library file, specify correct path
01-01 00:03:03.807 E/HAL     ( 1029): Could not find a HAL module for gralloc
01-01 00:03:03.807 D/NxpFwDnld( 1047): FW Image Length - ImageInfoLen 0
01-01 00:03:03.807 D/NxpFwDnld( 1047): FW Image Info Pointer - pImageInfo 0x0
01-01 00:03:03.807 D/NxpTml  ( 1047): PN54X - Invoking I2C Read.....
01-01 00:03:03.807 E/NxpFwDnld( 1047): Image extraction Failed - invalid imginfo or imginfolen!!
01-01 00:03:03.807 E/NxpFwDnld( 1047): Error loading libpn54x_fw !!
01-01 00:03:03.807 D/NxpHal  ( 1047): FW version for FW file = 0x0
01-01 00:03:03.808 D/NxpHal  ( 1047): FW version from device = 0x80202
01-01 00:03:03.808 D/NxpHal  ( 1047): phNxpNciHal_ioctl : exit - ret = -1
01-01 00:03:03.808 I/BrcmNfcNfa( 1047): fw_update required  -> 1
01-01 00:03:03.808 I/BrcmNfcNfa( 1047): uicc_eeprom_get_config : Enter
01-01 00:03:03.808 I/BrcmNfcNfa( 1047): NXP_DUAL_UICC_ENABLE not found; taking default value : 0x00
01-01 00:03:03.808 D/NfcAdaptation( 1047): NfcAdaptation::HalIoctl
01-01 00:03:03.808 D/NxpHal  ( 1047): phNxpNciHal_ioctl : enter - arg = 11
01-01 00:03:03.808 D/NxpTml  ( 1047): PN54X - Write requested.....
01-01 00:03:03.808 D/NxpTml  ( 1047): PN54X - Invoking I2C Write.....
01-01 00:03:03.809 I/OpenGLRenderer( 1029): Initialized EGL, version 1.4
01-01 00:03:03.810 D/NxpNciX ( 1047): len =   6 > 20030301A0EC
01-01 00:03:03.810 D/NxpTml  ( 1047): PN54X - I2C Write successful.....
01-01 00:03:03.810 D/NxpTml  ( 1047): PN54X - Posting Fresh Write message.....
01-01 00:03:03.810 D/NxpTml  ( 1047): PN54X - Tml Writer Thread Running................
01-01 00:03:03.810 D/NxpHal  ( 1047): write successful status = 0x0
01-01 00:03:03.810 D/NxpHal  ( 1047): Response timer started
01-01 00:03:03.810 D/NxpHal  ( 1047): Waiting after ext cmd sent
01-01 00:03:03.812 D/NxpTml  ( 1047): PN54X - I2C Read successful.....
01-01 00:03:03.812 D/NxpNciR ( 1047): len =   9 > 4003060001A0EC0100
01-01 00:03:03.812 D/NxpTml  ( 1047): PN54X - Posting read message.....
01-01 00:03:03.813 D/NxpHal  ( 1047): read successful status = 0x0
01-01 00:03:03.813 D/NxpHal  ( 1047): phNxpNciHal_print_res_status: response status =STATUS_OK
01-01 00:03:03.813 D/NxpHal  ( 1047): Response timer stopped
01-01 00:03:03.813 D/NxpHal  ( 1047): Checking response
01-01 00:03:03.813 D/NxpHal  ( 1047): phNxpNciHal_ioctl : exit - ret = 0
01-01 00:03:03.813 D/NfcAdaptation( 1047): NfcAdaptation::HalIoctl
01-01 00:03:03.813 D/NxpHal  ( 1047): phNxpNciHal_ioctl : enter - arg = 8
01-01 00:03:03.813 D/NxpHal  ( 1047): phNxpNciHal_CheckValidFwVersion current_major_no = 0x0
01-01 00:03:03.813 E/NxpHal  ( 1047): Wrong FW Version >>> Firmware download not allowed
01-01 00:03:03.813 D/NxpHal  ( 1047): phNxpNciHal_ioctl : exit - ret = -1
01-01 00:03:03.813 I/BrcmNfcNfa( 1047): Inside nfc_hal_nfcc_reset
01-01 00:03:03.813 D/NfcAdaptation( 1047): NfcAdaptation::HalIoctl
01-01 00:03:03.813 D/NxpHal  ( 1047): phNxpNciHal_ioctl : enter - arg = 11
[... 1200 lines cut ...]
01-01 00:03:04.632 E/BrcmNfcNfa( 1047): nfa_ce_api_dereg_listen (): cannot find listen_info for UICC/ESE
01-01 00:03:04.632 D/BrcmNfcJni( 1047): nfaConnectionCallback: event= 29
01-01 00:03:04.632 D/BrcmNfcJni( 1047): nfaConnectionCallback: NFA_CE_UICC_LISTEN_CONFIGURED_EVT : status=0x9
[... cut ...]
01-01 00:03:04.634 I/BrcmNfcNfa( 1047): Cannot find cb for given nfcee_id: 0x2
01-01 00:03:04.634 I/BrcmNfcNfa( 1047): supported tech list is 0x00 for given nfcee_id: 0x2 
01-01 00:03:04.634 I/BrcmNfcNfa( 1047): dm_disc_mask = 0x0
01-01 00:03:04.634 I/BrcmNfcNfa( 1047): nfa_dm_get_rf_discover_config : dm_disc_mask = 0x00000000
01-01 00:03:04.634 D/BrcmNfcJni( 1047): nfaConnectionCallback: event= 30
01-01 00:03:04.634 D/BrcmNfcJni( 1047): nfaConnectionCallback: NFA_RF_DISCOVERY_STARTED_EVT: status = 0

jgeslin, 2016-08-19:
Please consider recent PN7120/PN7150 support official delivery for Android Marshmallow [here](https://github.com/NXPNFCLinux/nxpnfc_android_marshmallow).

NicolasYJZ, 2016-09-23:
hi 
i'm updating android L to M, I have got the same problem of nfc ,i2c is ok,but couldn't enable NFC in settings   what do u mean "updated some code to make IIC command on M are same as Lollipop"

NicolasYJZ, 2016-09-23:
the log
09-23 01:35:22.225: D/NfcService(4331): NFC is on. Doing normal stuff
09-23 01:35:22.225: I/NfcService(4331): Enabling NFC
09-23 01:35:22.227: D/BrcmNfcJni(4331): nfcManager_getNfcInitTimeout: enter
09-23 01:35:22.227: D/BrcmNfcJni(4331): NAME_NXP_DEFAULT_NFCEE_DISC_TIMEOUT not found
09-23 01:35:22.227: D/BrcmNfcJni(4331):  gNfcInitTimeout = 6000: gdisc_timeout = 0 nfcManager_getNfcInitTimeout
09-23 01:35:22.227: I/NfcService(4331): Enabling NFC timeout90000
09-23 01:35:22.231: W/System(4449): ClassLoader referenced unknown path: /system/app/DeskClock/lib/x86
09-23 01:35:22.235: D/BrcmNfcJni(4331): nfcManager_doInitialize: enter; ver=NFCDROID-AOSP_L_00.01 nfa=NFA_PI_1.03.66+ NCI_VERSION=0x10
09-23 01:35:22.235: D/BrcmNfcJni(4331): nfcManager_doInitialize:  MW Version: NFC_NCIHALx_ARF.4.2.0
09-23 01:35:22.235: D/BrcmNfcJni(4331): PowerSwitch::initialize: level=PS-FULL (1)
09-23 01:35:22.235: D/NfcAdaptation(4331): const CNfcParam* CNfcConfig::find(const char*) const found SCREEN_OFF_POWER_STATE=(0x1)
09-23 01:35:22.235: D/BrcmNfcJni(4331): PowerSwitch::initialize: desired screen-off state=1
09-23 01:35:22.235: D/NfcAdaptation(4331): NfcAdaptation::Initialize: enter
09-23 01:35:22.235: E/NfcAdaptation(4331): NfcAdaptation::Initialize: ver=NFCDROID-AOSP_L_00.01 nfa=NFA_PI_1.03.66+
09-23 01:35:22.235: D/NfcAdaptation(4331): const CNfcParam* CNfcConfig::find(const char*) const found NFA_STORAGE=/data/nfc
09-23 01:35:22.235: D/NfcAdaptation(4331): const CNfcParam* CNfcConfig::find(const char*) const found PROTOCOL_TRACE_LEVEL=(0xFFFFFFFF)
09-23 01:35:22.235: D/NfcAdaptation(4331): const CNfcParam* CNfcConfig::find(const char*) const found NFA_MAX_EE_SUPPORTED=(0x3)
09-23 01:35:22.235: D/NfcAdaptation(4331): NfcAdaptation::Initialize: Overriding NFA_EE_MAX_EE_SUPPORTED to use 3
09-23 01:35:22.235: D/NfcAdaptation(4331): const CNfcParam* CNfcConfig::find(const char*) const found APPL_TRACE_LEVEL=(0xFF)
09-23 01:35:22.235: D/BrcmNfcJni(4331): initializeGlobalAppLogLevel: level=5
09-23 01:35:22.235: D/(4331): verify_stack_non_volatile_store
09-23 01:35:22.235: D/NfcAdaptation(4331): Initialize: preserve stack NV store
09-23 01:35:22.236: D/NfcAdaptation(4331): NfcAdaptation::NFCA_TASK: enter
09-23 01:35:22.236: I/BrcmNfcNfa(4331): GKI_run(): Start/Stop GKI_timer_update_registered!
09-23 01:35:22.238: E/tm_logprobe(3194): Unable to create record: No data available
09-23 01:35:22.239: D/NfcAdaptation(4331): NfcAdaptation::Thread: enter
09-23 01:35:22.242: I/BrcmNfcNfa(4331): NFC_TASK started.
09-23 01:35:22.242: D/NfcAdaptation(4331): NfcAdaptation::Thread: exit
09-23 01:35:22.242: D/NfcAdaptation(4331): NfcAdaptation::InitializeHalDeviceContext: enter
09-23 01:35:22.242: D/NfcAdaptation(4331): const CNfcParam* CNfcConfig::find(const char*) const found NCI_HAL_MODULE=nfc_nci.pn54x
09-23 01:35:22.242: I/HAL client(4331): HAL_IPC_MAP failed: No such file or directory
09-23 01:35:22.242: E/HAL(4331): Could not find a HAL module for nfc_nci.pn54x
09-23 01:35:22.249: D/NxpNfcNciHal(4331): nfc_open: enter; name=nci
09-23 01:35:22.249: D/NxpNfcNciHal(4331): nfc_open: exit 0
09-23 01:35:22.249: D/NfcAdaptation(4331): NfcAdaptation::InitializeHalDeviceContext: exit
09-23 01:35:22.249: D/NfcAdaptation(4331): NfcAdaptation::Initialize: exit
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_dm_init ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_sys_register () id=1, enable_cplt_mask=0x0
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_sys_register () id=3, enable_cplt_mask=0x8
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_rw_init ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_sys_register () id=6, enable_cplt_mask=0x48
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_ce_init ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_sys_register () id=7, enable_cplt_mask=0xc8
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_ee_init ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_sys_register () id=2, enable_cplt_mask=0xcc
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_hci_init ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_sys_register () id=8, enable_cplt_mask=0x1cc
09-23 01:35:22.250: I/BrcmNfcNfa(4331): LLCP - llcp_init ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): num_rx_buff = 21, rx_congest_start = 14, rx_congest_end = 10, max_num_ll_rx_buff = 6
09-23 01:35:22.250: I/BrcmNfcNfa(4331): max_num_tx_buff = 49, max_num_ll_tx_buff = 14
09-23 01:35:22.250: I/BrcmNfcNfa(4331): LLCP_RegisterServer (): SAP:0x1, link_type:0x2, ServiceName:<urn:nfc:sn:sdp>
09-23 01:35:22.250: I/BrcmNfcNfa(4331): LLCP_RegisterServer (): Registered SAP = 0x01
09-23 01:35:22.250: I/BrcmNfcNfa(4331): NFA_Enable ()
09-23 01:35:22.250: D/NfcAdaptation(4331): const CNfcParam* CNfcConfig::find(const char*) const found APPL_TRACE_LEVEL=(0xFF)
09-23 01:35:22.250: I/BrcmNfcNfa(4331): NFA got event 0x0100
09-23 01:35:22.250: D/BrcmNfcJni(4331): initializeGlobalAppLogLevel: level=5
09-23 01:35:22.250: I/BrcmNfcNfa(4331): NFC_SetTraceLevel () new_level = 5
09-23 01:35:22.250: I/BrcmNfcNfa(4331): NFC_SetTraceLevel () new_level = 5
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfa_dm_enable ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): NFC_Enable ()
09-23 01:35:22.250: I/BrcmNfcNfa(4331): nfc_set_state 0 (NONE)->1 (W4_HAL_OPEN)
09-23 01:35:22.250: D/NfcAdaptation(4331): NfcAdaptation::HalOpen
09-23 01:35:22.258: D/NxpExtns(4331): const nxp::CNfcParam* nxp::CNfcConfig::find(const char*) const found NXPLOG_TML_LOGLEVEL=(0x3)
[... cut ...]
09-23 01:35:22.815: D/NxpTml(4331): PN54X - Tml Reader Thread Started................
09-23 01:35:22.815: D/NxpTml(4331): PN54X - Read requested.....
09-23 01:35:22.815: D/NxpTml(4331): PN54X - Invoking I2C Read.....
09-23 01:35:22.824: E/NxpTml(4331): _i2c_write() errno : 79
09-23 01:35:22.824: E/NxpTml(4331): PN54X - Error in I2C Write.....
09-23 01:35:22.824: D/NxpTml(4331): PN54X - Posting Fresh Write message.....
09-23 01:35:22.824: E/NxpHal(4331): write error status = 0x1ff
09-23 01:35:22.824: D/NxpTml(4331): PN54X - Tml Writer Thread Running................

jgeslin, 2016-09-23:
Hi NicolasYJZ,

Seems you have an issue with the kernel driver, I2C write operation fails:
>  09-23 01:35:22.824: E/NxpTml(4331): _i2c_write() errno : 79

Please check the kernel message using "dmesg" command.

Regards,
Jeremy.

NicolasYJZ, 2016-09-23:
mtp_open
[   33.937518] pn544_platform_init : moofu sending reset cmd
[   33.954606] pn544_platform_init : moofu ret = -121
[   33.993044] pn544_platform_init : moofu sending reset cmd
[   34.000989] pn544_platform_init : moofu ret = -121
[   34.037092] pn544_platform_init : moofu sending reset cmd
[   34.044708] pn544_platform_init : moofu ret = -121
[   34.082150] pn544_platform_init : moofu sending reset cmd
[   34.091369] pn544_platform_init : moofu ret = -121
[   34.127207] pn544_platform_init : moofu sending reset cmd
[   34.137027] pn544_platform_init : moofu ret = -121
[   34.178274] pn544_platform_init : moofu sending reset cmd
[   34.185506] pn544_platform_init : moofu ret = -121
[   34.190941] pn544_platform_init : moonfu could not detect nfc_en polarity
[   34.198800] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[   34.321609] pn544_set_power moonfu power off
[   34.442129] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[   34.469397] pn544_dev_write : moonfu i2c_master_send returned -121
[   34.478347] pn544_dev_write : moonfu i2c_master_send returned -121
[   34.487735] pn544_dev_write : moonfu i2c_master_send returned -121
[   34.498713] pn544_dev_write : moonfu i2c_master_send returned -121
[   34.507341] pn544_dev_write : moonfu i2c_master_send returned -121
[   34.524935] pn544_dev_write : moonfu i2c_master_send returned -121
[   34.532288] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[   34.655007] pn544_set_power moonfu power off
[   34.775524] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[   34.798352] pn544_set_power moonfu power on with firmware
[   34.950486] pn544_set_power moonfu power off
[   34.980506] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[   35.666814] init: Service 'bootanim' (pid 2747) exited with status 0
[   35.827505] init: write_file: Unable to open '/sys/block/dm-1/queue/read_ahead_kb': No such file or directory
[   71.118507]  HAL: libhal_tm_post_processing: Could not allocate telemetry record
[  124.578448] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  124.701934] pn544_set_power moonfu power off
[  124.822676] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  124.859365] pn544_dev_write : moonfu i2c_master_send returned -121
[  124.869291] pn544_dev_write : moonfu i2c_master_send returned -121
[  124.879598] pn544_dev_write : moonfu i2c_master_send returned -121
[  124.889121] pn544_dev_write : moonfu i2c_master_send returned -121
[  124.898358] pn544_dev_write : moonfu i2c_master_send returned -121
[  124.907239] pn544_dev_write : moonfu i2c_master_send returned -121
[  124.915618] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  125.034631] pn544_set_power moonfu power off
[  125.156281] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  125.180407] pn544_set_power moonfu power on with firmware
[  125.334228] pn544_set_power moonfu power off
[  125.365625] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  177.724212] [drm] Enabling RC6 states: RC6 on, RC6p off, RC6pp off
[  178.060563] [drm:intel_set_disp_calc_flip] *ERROR* pfit mode(268435456) mismatch with calculated mode(0)
[  215.235114] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  215.359182] pn544_set_power moonfu power off
[  215.480197] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  215.516127] pn544_dev_write : moonfu i2c_master_send returned -121
[  215.526131] pn544_dev_write : moonfu i2c_master_send returned -121
[  215.535425] pn544_dev_write : moonfu i2c_master_send returned -121
[  215.545048] pn544_dev_write : moonfu i2c_master_send returned -121
[  215.554461] pn544_dev_write : moonfu i2c_master_send returned -121
[  215.563460] pn544_dev_write : moonfu i2c_master_send returned -121
[  215.570717] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  215.694655] pn544_set_power moonfu power off
[  215.815264] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  215.839308] pn544_set_power moonfu power on with firmware
[  215.992562] pn544_set_power moonfu power off
[  216.022321] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  256.820599] [drm] Enabling RC6 states: RC6 on, RC6p off, RC6pp off
[  305.982024] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  306.105505] pn544_set_power moonfu power off
[  306.226446] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  306.258440] pn544_dev_write : moonfu i2c_master_send returned -121
[  306.267723] pn544_dev_write : moonfu i2c_master_send returned -121
[... 13 lines cut ...]
[  396.753133] pn544_set_power moonfu power off
[  396.873548] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  396.909763] pn544_dev_write : moonfu i2c_master_send returned -121
[  396.920339] pn544_dev_write : moonfu i2c_master_send returned -121
[  396.930703] pn544_dev_write : moonfu i2c_master_send returned -121
[  396.940841] pn544_dev_write : moonfu i2c_master_send returned -121
[  396.950490] pn544_dev_write : moonfu i2c_master_send returned -121
[  396.960626] pn544_dev_write : moonfu i2c_master_send returned -121
[  396.968399] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  397.091757] pn544_set_power moonfu power off
[  397.213073] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[... cut ...]
[  487.574575] pn544_dev_write : moonfu i2c_master_send returned -121
[  487.583990] pn544_dev_write : moonfu i2c_master_send returned -121
[  487.593113] pn544_dev_write : moonfu i2c_master_send returned -121
[  487.602857] pn544_dev_write : moonfu i2c_master_send returned -121
[  487.612318] pn544_dev_write : moonfu i2c_master_send returned -121
[  487.622092] pn544_dev_write : moonfu i2c_master_send returned -121
[  487.629638] pn544_set_power moonfu power on pn544_dev->nfc_en_polarity = 2
[  487.748802] pn544_set_power moonfu power off

NicolasYJZ, 2016-09-23:
what dose this mean?   the driver have problems?

jgeslin, 2016-09-23:
Error 121 means:
> #define EREMOTEIO   121 /* Remote I/O error */

It can be due to I2C address not acknowledged by the PN71xx.
Please check VEN signal is high and verify I2C lines.

NicolasYJZ, 2016-09-24:
Thank you  i will check it

jgeslin, 2016-10-18:
Can we consider this item closed ?

closed 2016-12-26 by wintonliuwen