NXPNFCLinux/nxp-pn5xx #9: dev/pn544 registration issue on Android Lollipop

Jameskody, 2017-03-06:
Hi

Trying to use NXP NFC PN544 on Android lollipop version 5.1.1

Referring to the AN11690 NXPNCI Android Porting Guidelines.pdf document,
The porting was done in the following order.
1. Download Driver
   $ cd drivers/misc 
   $ git clone https://github.com/NXPNFCLinux/nxp-pn5xx.git
   
2. Kernel modification
   drivers/misc/Makefile :Add to file
   obj-y += nxp-pn5xx/
   
   drivers/misc/Kconfig :Add to file
   source "drivers/misc/nxp-pn5xx/Kconfig"
   
3. Device node setup
   arch/arm/boot/dts/RK3128_fireprime.dts (ARM based platform):Add to file
   
   &i2c{ 
      status = "okay";
      pn544@54 { 
      compatible = "nxp,pn547"; 
      reg = <0x54>; 
      clock-frequency = <400000>; 
      interrupt-gpios = <&gpio3 GPIO_C4 IRQ_TYPE_EDGE_FALLING>; 
      enable-gpios = <&gpio1 GPIO_C7 GPIO_ACTIVE_LOW>; 
      }; 
   };

4. Build driver
   make menuconfig :Select after execution
   
   Device Drivers ---> 
        Misc devices ---> 
           <*> NXP PN5XX based driver
   
   Kernel build succeeded.

5. NXP's Android lollipop NFC delivery
   $ curl https://raw.githubusercontent.com/NXPNFCLinux/nxpnfc_android_lollipop/master/nxpnfc_manifest.xml > nxpnfc_manifest.xml
   
6. NXP-NCI delivery
   $ ./NxpNfcAndroid/install.sh PN7120
   
7. Adding an NFC build
   device/brand/platform/device.mk :Add to file
   # NFC packages 
        PRODUCT_PACKAGES += \ 
             libnfc-nci \ 
             libnfc_nci_jni \ 
             nfc_nci_pn54x.default\ 
             NfcNci \ 
             Tag \ 
             com.android.nfc_extras 
        
        PRODUCT_COPY_FILES += \               frameworks/native/data/etc/com.nxp.mifare.xml:system/etc/permissions/com.nxp.mifare.xml\               frameworks/native/data/etc/com.android.nfc_extras.xml:system/etc/permissions/com.android.nfc_extras.xml \             frameworks/native/data/etc/android.hardware.nfc.xml:system/etc/permissions/android.hardware.nfc.xml \              frameworks/native/data/etc/android.hardware.nfc.hce.xml:system/etc/permissions/android.hardware.nfc.hce.xml \ 
NxpNfcAndroid/conf/libnfc-brcm.conf:system/etc/libnfc-brcm.conf \ 
NxpNfcAndroid/conf/libnfc-nxp.conf:system/etc/libnfc-nxp.conf

8. Permission Settings
   system/core/rootdir/init.rc :Add to file
   # NFC
     setprop ro.nfc.port "I2C" 
     chmod 0660 /dev/pn544 
     chown nfc nfc /dev/pn544
     
9. System build successful
   Build and flash the system image.
  -. From the terminal after booting the system 
     $adb shell ls -als /dev/pn544 
      /dev/pn544: No such file or directory   

  -. After booting the system, in the setting window of Android screen
     In “Settings” app check NFC is ON. (Can not be turned on or off) 
   
 Inde above situation, what part do you need to make to be registered in / dev / pn544.  

 Regards.

jgeslin, 2017-03-06:
Hi,

If the /dev/pn544 device is not visible it means the kernel driver is not loaded properly at boot time.
It can be several root causes (driver not included in kernel image, device not properly referenced in device tree, ...).
Looking to the kernel logs (using "dmesg" command) should help identifying the root cause.

Regards,
Jeremy.

Jameskody, 2017-03-22:
Hi,
I confirmed the registration of / dev / pn544 in the ADB shell.

An error occurred in the kernel log as shown below.
"pn54x_dev_release: closing 10,57" is what appears as normal?

6>[    2.073333] pn54x_dev_init
<6>[    2.073438] pn54x_probe
<7>[    2.073494] of_get_named_gpio_flags exited with status 65
<7>[    2.073511] of_get_named_gpio_flags: can't parse gpios property
<4>[    2.073535] pn544 2-0050: FIRM GPIO <OPTIONAL> error getting from OF node
<7>[    2.073565] of_get_named_gpio_flags exited with status 89
<7>[    2.073581] of_get_named_gpio_flags: can't parse gpios property
<4>[    2.073602] pn544 2-0050: CLKREQ GPIO <OPTIONAL> error getting from OF node
<3>[    2.073638] pn54x_get_pdata: could not get nxp,pn54x-pvdd, rc=-19
<3>[    2.073663] pn54x_get_pdata: could not get nxp,pn54x-vbat, rc=-19
<3>[    2.073688] pn54x_get_pdata: could not get nxp,pn54x-pmuvcc, rc=-19
<3>[    2.073712] pn54x_get_pdata: could not get nxp,pn54x-sevdd, rc=-19
<6>[    2.073734] pn54x_probe: request irq_gpio 89
<6>[    2.073817] pn54x_probe: request ven_gpio 65
<6>[    2.074113] pn54x_probe : requesting IRQ 165

<6>[    4.573281] pn54x_dev_open : 10,57
<6>[    4.573320] pn54x_dev_release : closing 10,57
<6>[    4.573365] pn54x_dev_open : 10,57
<6>[    4.573393] pn54x_dev_release : closing 10,57
<6>[    4.578657] binder: 137:137 transaction failed 29189, size 0-0
<7>[    4.591355] _gpio_request: gpio-65 (sysfs) status -16
<7>[    4.591377] export_store: status -16
<7>[    4.593947] _gpio_request: gpio-89 (sysfs) status -16
<7>[    4.593969] export_store: status -16

jgeslin, 2017-03-22:
Hi,

The dev_open/dev_release kernel traces relates to driver access from the User space (probably from the NFC service). Do you see the NFC service starting properly in the ADB logs ?

Regards,
Jeremy.

Jameskody, 2017-03-22:
Hi,
Thank you for your quick reply.

The logcat message from the ADB shell below.

D/BrcmNfcJni(  804): JNI_OnLoad: enter
I/BrcmNfcJni(  804): NFC Service: loading nci JNI
D/BrcmNfcJni(  804): register_com_android_nfc_NativeNfcManager: enter
D/BrcmNfcJni(  804): PowerSwitch::initialize: level=PS-UNKNOWN (0)
D/NfcAdaptation(  804): bool CNfcConfig::readConfig(const char*, bool) Opened base config /etc/libnfc-brcm.conf
D/NfcAdaptation(  804): const CNfcParam* CNfcConfig::find(const char*) const found SCREEN_OFF_POWER_STATE=(0x1)
D/BrcmNfcJni(  804): PowerSwitch::initialize: desired screen-off state=1
D/BrcmNfcJni(  804): register_com_android_nfc_NativeNfcManager: exit
D/BrcmNfcJni(  804): register_com_android_nfc_NativeNfcTag
D/BrcmNfcJni(  804): RoutingManager::RoutingManager(): default route is 0x00
D/BrcmNfcJni(  804): RoutingManager::RoutingManager(): mOffHostEe=0x02 mAidMatchingMode=0x 0
D/BrcmNfcJni(  804): RoutingManager::registerJniFunctions
D/BrcmNfcJni(  804): JNI_OnLoad: exit
D/Launcher.LauncherProvider(  842): 11683562 - initializeMaxScreenId(): 3
D/BrcmNfcJni(  804): nfcManager_initNativeStruc: enter
D/BrcmNfcJni(  804): nfcManager_initNativeStruc: exit
V/Launcher(  842): LauncherAppState inited
W/NfceeAccess(  804): could not find /etc/nfcee_access.xml, no NFCEE access allowed
I/NfceeAccess(  804): read 0 signature(s) for NFCEE access
W/NfcSccAccess(  804): could not find /etc/nfcscc_access.xml, no NFCSCC access allowed
I/NfcSccAccess(  804): read 0 signature(s) for NFCSCC access
W/NfcSeAccess(  804): could not find /etc/nfcse_access.xml, no NFCSE access allowed
I/NfcSeAccess(  804): read 0 signature(s) for NFCSE access
D/NfcService(  804): Enforcing a policy change on user: UserHandle{0}, isActiveForUser = true
D/BluetoothManagerService(  420): Message: 300
D/BluetoothManagerService(  420): MESSAGE_USER_SWITCHED
W/Settings(  804): Setting airplane_mode_radios has moved from android.provider.Settings.System to android.provider.Settings.Global, returning read-only value.
W/Settings(  804): Setting airplane_mode_toggleable_radios has moved from android.provider.Settings.System to android.provider.Settings.Global, returning read-only value.
D/PermissionCache(  141): checking android.permission.HARDWARE_TEST for uid=1000 => granted (257 us)
D/AidRoutingManager(  804): mDefaultRoute=0x0
D/AidRoutingManager(  804): mDefaultOffHostRoute=0x2
D/AidRoutingManager(  804): mAidMatchingSupport=0x0
D/AidRoutingManager(  804): mAidMatchingPlatform=0x1
D/AidRoutingManager(  804): mAidTableSize=0xa0
D/CardEmulationManager(  804): Current default: null
D/CardEmulationManager(  804): No default set, last payment service removed.
D/RegisteredAidCache(  804): onServicesUpdated
D/RegisteredAidCache(  804): Not updating routing table because NFC is off.
D/RegisteredNfcid2Cache(  804): onServicesUpdated
D/RegisteredNfcid2Cache(  804): generateAidCacheLocked: mNfcid2Cache size 0
D/RegisteredNfcid2Cache(  804): Not updating routing table because NFC is off.
D/NfcService(  804): Before mIsHceCapable
D/NfcService(  804): checking on firmware download
W/Settings(  804): Setting airplane_mode_on has moved from android.provider.Settings.System to android.provider.Settings.Global, returning read-only value.
D/NfcService(  804): NFC is on. Doing normal stuff
I/NfcService(  804): Enabling NFC
D/BrcmNfcJni(  804): nfcManager_getNfcInitTimeout: enter
D/BrcmNfcJni(  804): NAME_NXP_DEFAULT_NFCEE_DISC_TIMEOUT not found
D/BrcmNfcJni(  804): NAME_NXP_DEFAULT_NFCEE_TIMEOUT not found
D/BrcmNfcJni(  804):  gNfcInitTimeout = 0: gdisc_timeout = 0 nfcManager_getNfcInitTimeout
I/NfcService(  804): Enabling NFC timeout90000
I/FusedLocation(  420): engine started (com.android.location.fused)
D/BrcmNfcJni(  804): nfcManager_doInitialize: enter; ver=NFCDROID-AOSP_L_00.01 nfa=NFA_PI_1.03.66+ NCI_VERSION=0x10
D/BrcmNfcJni(  804): nfcManager_doInitialize:  MW Version: NFC_NCIHALx_ARF.3.5.0
D/BrcmNfcJni(  804): PowerSwitch::initialize: level=PS-FULL (1)
D/NfcAdaptation(  804): const CNfcParam* CNfcConfig::find(const char*) const found SCREEN_OFF_POWER_STATE=(0x1)
D/BrcmNfcJni(  804): PowerSwitch::initialize: desired screen-off state=1
D/NfcAdaptation(  804): NfcAdaptation::Initialize: enter
E/NfcAdaptation(  804): NfcAdaptation::Initialize: ver=NFCDROID-AOSP_L_00.01 nfa=NFA_PI_1.03.66+
D/NfcAdaptation(  804): const CNfcParam* CNfcConfig::find(const char*) const found PROTOCOL_TRACE_LEVEL=(0xFFFFFFFF)
D/NfcAdaptation(  804): const CNfcParam* CNfcConfig::find(const char*) const found NFA_MAX_EE_SUPPORTED=(0x1)
D/NfcAdaptation(  804): NfcAdaptation::Initialize: Overriding NFA_EE_MAX_EE_SUPPORTED to use 1
D/NfcAdaptation(  804): const CNfcParam* CNfcConfig::find(const char*) const found APPL_TRACE_LEVEL=(0xFF)
D/BrcmNfcJni(  804): initializeGlobalAppLogLevel: level=5
D/        (  804): verify_stack_non_volatile_store
[... 858 lines cut ...]
I/BrcmNfcNfa( 1476): nfa_rw_init ()
I/BrcmNfcNfa( 1476): nfa_sys_register () id=6, enable_cplt_mask=0x48
I/BrcmNfcNfa( 1476): nfa_ce_init ()
I/BrcmNfcNfa( 1476): nfa_sys_register () id=7, enable_cplt_mask=0xc8
I/BrcmNfcNfa( 1476): nfa_ee_init ()
I/BrcmNfcNfa( 1476): nfa_sys_register () id=2, enable_cplt_mask=0xcc
I/BrcmNfcNfa( 1476): nfa_hci_init ()
I/BrcmNfcNfa( 1476): nfa_sys_register () id=8, enable_cplt_mask=0x1cc
I/BrcmNfcNfa( 1476): LLCP - llcp_init ()
I/BrcmNfcNfa( 1476): num_rx_buff = 21, rx_congest_start = 14, rx_congest_end = 10, max_num_ll_rx_buff = 6
[... cut ...]
I/BrcmNfcNfa( 1476): NFA got event 0x0100
D/BrcmNfcJni( 1476): initializeGlobalAppLogLevel: level=5
I/BrcmNfcNfa( 1476): nfa_dm_evt_hdlr event: NFA_DM_API_ENABLE_EVT (0x00)
I/BrcmNfcNfa( 1476): NFC_SetTraceLevel () new_level = 5
I/BrcmNfcNfa( 1476): nfa_dm_enable ()
I/BrcmNfcNfa( 1476): NFC_Enable ()
I/BrcmNfcNfa( 1476): nfc_set_state 0 (NONE)->1 (W4_HAL_OPEN)
D/NfcAdaptation( 1476): NfcAdaptation::HalOpen
E/NfcService( 1476): Watchdog triggered, aborting.
E/BrcmNfcJni( 1476): nfcManager_doAbort: abort()

Jameskody, 2017-03-24:
Hi,
In the message below "pn54x_dev_open: 10,57"
10, 57 means pn547?
It is pn544 to test now. Is it irrelevant?

shell@dvx1k:/ $ dmesg 
6>[ 2.073333] pn54x_dev_init
 <6>[ 2.073438] pn54x_probe
 <7>[ 2.073494] of_get_named_gpio_flags exited with status 65
 <7>[ 2.073511] of_get_named_gpio_flags: can't parse gpios property
 <4>[ 2.073535] pn544 2-0050: FIRM GPIO error getting from OF node
 <7>[ 2.073565] of_get_named_gpio_flags exited with status 89
 <7>[ 2.073581] of_get_named_gpio_flags: can't parse gpios property
 <4>[ 2.073602] pn544 2-0050: CLKREQ GPIO error getting from OF node
 <3>[ 2.073638] pn54x_get_pdata: could not get nxp,pn54x-pvdd, rc=-19
 <3>[ 2.073663] pn54x_get_pdata: could not get nxp,pn54x-vbat, rc=-19
 <3>[ 2.073688] pn54x_get_pdata: could not get nxp,pn54x-pmuvcc, rc=-19
 <3>[ 2.073712] pn54x_get_pdata: could not get nxp,pn54x-sevdd, rc=-19
 <6>[ 2.073734] pn54x_probe: request irq_gpio 89
 <6>[ 2.073817] pn54x_probe: request ven_gpio 65
 <6>[ 2.074113] pn54x_probe : requesting IRQ 165
----Skip messages.----
<6>[ 4.573281] pn54x_dev_open : 10,57
 <6>[ 4.573320] pn54x_dev_release : closing 10,57
 <6>[ 4.573365] pn54x_dev_open : 10,57
 <6>[ 4.573393] pn54x_dev_release : closing 10,57
----Skip messages.----
<6>[  194.278133] pn544_enable power on
<6>[  194.481878] pn54x_dev_ioctl, cmd=1074063617, arg=0
<6>[  194.481982] pn544_disable power off
<6>[  194.688525] pn54x_dev_ioctl, cmd=1074063617, arg=1
<6>[  194.688632] pn544_enable power on
<6>[  194.799485] pn54x_dev_ioctl, cmd=1074063617, arg=2
<6>[  194.799593] pn544_enable power on with firmware
<6>[  195.093761] pn54x_dev_ioctl, cmd=1074063617, arg=0
<6>[  195.093874] pn544_disable power off
<6>[  195.210089] pn54x_dev_ioctl, cmd=1074063617, arg=1
<6>[  195.210203] pn544_enable power on
<6>[  282.309404] pn54x_dev_release : closing 10,57
<6>[  282.567265] pn54x_dev_open : 10,57
<6>[  282.567349] pn54x_dev_ioctl, cmd=1074063617, arg=1
<6>[  282.567364] pn544_enable power on
<6>[  282.766312] pn54x_dev_ioctl, cmd=1074063617, arg=0
<6>[  282.766418] pn544_disable power off
<6>[  282.973018] pn54x_dev_ioctl, cmd=1074063617, arg=1
<6>[  282.973119] pn544_enable power on
<3>[  283.085180] pn54x_dev_write : i2c_master_send returned -11
<3>[  283.090772] pn54x_dev_write : i2c_master_send returned -11
<3>[  283.095928] pn54x_dev_write : i2c_master_send returned -11
<3>[  283.101226] pn54x_dev_write : i2c_master_send returned -11
<3>[  283.106412] pn54x_dev_write : i2c_master_send returned -11
<3>[  283.111564] pn54x_dev_write : i2c_master_send returned -11
<6>[  283.113887] pn54x_dev_ioctl, cmd=1074063617, arg=1
<6>[  283.113959] pn544_enable power on
<6>[  283.317422] pn54x_dev_ioctl, cmd=1074063617, arg=0
<6>[  283.317529] pn544_disable power off
<6>[  283.523946] pn54x_dev_ioctl, cmd=1074063617, arg=1
<6>[  283.524047] pn544_enable power on
<6>[  283.634505] pn54x_dev_ioctl, cmd=1074063617, arg=2
<6>[  283.634561] pn544_enable power on with firmware
<6>[  283.929289] pn54x_dev_ioctl, cmd=1074063617, arg=0
<6>[  283.929405] pn544_disable power off
<6>[  284.045520] pn54x_dev_ioctl, cmd=1074063617, arg=1
<6>[  284.045622] pn544_enable power on
 -----After of 5 seconds----------
 [  371.144804] pn54x_dev_release : closing 10,57
Then, repeat from pn544_enable power on to pn544_disable power off.
Do you know what causes this?

Regards,
 James.

Jameskody, 2017-03-24:
The following is a message to check the above with the logcat command.

shell@dvx1k:/ $ logcat
----Skip messages.----
I/NfcService( 1858): Starting NFC service
D/BrcmNfcJni( 1858): JNI_OnLoad: enter
I/BrcmNfcJni( 1858): NFC Service: loading nci JNI
D/BrcmNfcJni( 1858): register_com_android_nfc_NativeNfcManager: enter
D/BrcmNfcJni( 1858): PowerSwitch::initialize: level=PS-UNKNOWN (0)
D/NfcAdaptation( 1858): bool CNfcConfig::readConfig(const char*, bool) Opened base config /etc/libnfc-brcm.conf
D/NfcAdaptation( 1858): const CNfcParam* CNfcConfig::find(const char*) const found SCREEN_OFF_POWER_STATE=(0x1)
D/BrcmNfcJni( 1858): PowerSwitch::initialize: desired screen-off state=1
D/BrcmNfcJni( 1858): register_com_android_nfc_NativeNfcManager: exit
D/BrcmNfcJni( 1858): register_com_android_nfc_NativeNfcTag
D/BrcmNfcJni( 1858): RoutingManager::RoutingManager(): default route is 0x00
D/BrcmNfcJni( 1858): RoutingManager::RoutingManager(): mOffHostEe=0x02 mAidMatchingMode=0x 0
D/BrcmNfcJni( 1858): RoutingManager::registerJniFunctions
D/BrcmNfcJni( 1858): JNI_OnLoad: exit
D/BrcmNfcJni( 1858): nfcManager_initNativeStruc: enter
D/BrcmNfcJni( 1858): nfcManager_initNativeStruc: exit
W/Settings( 1858): Setting device_provisioned has moved from android.provider.Settings.Secure to android.provider.Settings.Global.
W/NfceeAccess( 1858): could not find /etc/nfcee_access.xml, no NFCEE access allowed
I/NfceeAccess( 1858): read 0 signature(s) for NFCEE access
W/NfcSccAccess( 1858): could not find /etc/nfcscc_access.xml, no NFCSCC access allowed
I/NfcSccAccess( 1858): read 0 signature(s) for NFCSCC access
W/NfcSeAccess( 1858): could not find /etc/nfcse_access.xml, no NFCSE access allowed
I/NfcSeAccess( 1858): read 0 signature(s) for NFCSE access
D/NfcService( 1858): Enforci[ 1a3p.08cy9chapne x_ uveo: ns: Ha,57
[}, 3sA0t4291orps54x_ trueo
g[airp8an84m0de ra54o_ nas eoved from android.provider.Settings.System to android.provider.Settings.Global, returning read-only value.
W/Settings( 1858): Setting airplane_mode_toggleable_radios has moved from android.provider.Settings.System to android.provider.Settin
gs.Global, returning read-only value.
D/AidRoutingManager( 1858): mDefaultRoute=0x0
D/AidRoutingManager( 1858): mDefaultOffHostRoute=0x2
D/AidRoutingManager( 1858): mAidMatchingSupport=0x0
D/AidRoutingManager( 1858): mAidMatchingPlatform=0x1
D/AidRoutingManager( 1858): mAidTableSize=0xa0
D/RegisteredServicesCache( 1858): Dynamic AIDs file does not exist.
D/RegisteredNxpServicesCache( 1858): readDynamicApduService End:   0
D/CardEmulationManager( 1858): Current default: null
D/CardEmulationManager( 1858): No default set, last payment service removed.
D/RegisteredAidCache( 1858): onServicesUpdated
D/RegisteredAidCache( 1858): Not updating routing table because NFC is off.
D/RegisteredNfcid2Cache( 1858): onServicesUpdated
D/RegisteredNfcid2Cache( 1858): generateAidCacheLocked: mNfcid2Cache size 0
D/RegisteredNfcid2Cache( 1858): Not updating routing table because NFC is off.
D/NfcService( 1858): Before mIsHceCapable
D/NfcService( 1858): checking on firmware download
W/Settings( 1858): Setting airplane_mode_on has moved from android.provider.Settings.System to android.provider.Settings.Global, retu
rning read-only value.
D/NfcService( 1858): NFC is on. Doing normal stuff
I/NfcService( 1858): Enabling NFC
D/BrcmNfcJni( 1858): nfcManager_getNfcInitTimeout: enter
D/BrcmNfcJni( 1858): NAME_NXP_DEFAULT_NFCEE_DISC_TIMEOUT not found
D/BrcmNfcJni( 1858): NAME_NXP_DEFAULT_NFCEE_TIMEOUT not found
D/BrcmNfcJni( 1858):  gNfcInitTimeout = 0: gdisc_timeout = 0 nfcManager_getNfcInitTimeout
I/NfcService( 1858): Enabling NFC timeout90000
D/BrcmNfcJni( 1858): nfcManager_doInitialize: enter; ver=NFCDROID-AOSP_L_00.01 nfa=NFA_PI_1.03.66+ NCI_VERSION=0x10
D/BrcmNfcJni( 1858): nfcManager_doInitialize:  MW Version: NFC_NCIHALx_ARF.3.5.0
D/BrcmNfcJni( 1858): PowerSwitch::initialize: level=PS-FULL (1)
D/NfcAdaptation( 1858): const CNfcParam* CNfcConfig::find(const char*) const found SCREEN_OFF_POWER_STATE=(0x1)
D/BrcmNfcJni( 1858): PowerSwitch::initialize: desired screen-off state=1
D/NfcAdaptation( 1858): NfcAdaptation::Initialize: enter
E/NfcAdaptation( 1858): NfcAdaptation::Initialize: ver=NFCDROID-AOSP_L_00.01 nfa=NFA_PI_1.03.66+
D/NfcAdaptation( 1858): const CNfcParam* CNfcConfig::find(const char*) const found PROTOCOL_TRACE_LEVEL=(0xFFFFFFFF)
D/NfcAdaptation( 1858): const CNfcParam* CNfcConfig::find(const char*) const found NFA_MAX_EE_SUPPORTED=(0x1)
D/NfcAdaptation( 1858): NfcAdaptation::Initialize: Overriding NFA_EE_MAX_EE_SUPPORTED to use 1
D/NfcAdaptation( 1858): const CNfcParam* CNfcConfig::find(const char*) const found APPL_TRACE_LEVEL=(0xFF)
D/BrcmNfcJni( 1858): initializeGlobalAppLogLevel: level=5
D/        ( 1858): verify_stack_non_volatile_store
[... 150 lines cut ...]
 )1439ll8ng31e pn5n4_eisabr FW wer lof
D/NxpFwDnld( 1858): @@@/system/vendor/firmware/libpn547_fw.so
E/NxpFwDnld( 1858): NULL handler : unable to load the library file, specify correct path
D/NxpFwDnld( 1858): FW Image Length - ImageInfoLen 0
D/NxpFwDnld( 1858): FW Image Info Pointer - pImageInfo 0
E/NxpFwDnld( 1858): Image extraction Failed - invalid imginfo or imginfolen!!
E/NxpFwDnld( 1858): Error loading libpn54x_fw !!
E/NxpFwDnld( 1858): phDnldNfc_InitImgInfo: FAILED
D/NxpFwDnld( 1858): phNxpNciHal_fw_dnld_complete: Download Status = 0xff
E/NxpFwDnld( 1858): Fw Download Failed!!
D/NxpTml  ( 1858): phTmlNfc_i2c_reset(), VEN level 0
[... cut ...]
[ 1D/9x5T7l1 ] p8544_ nhbmeNfowircore
                                     et(), VEN level 1
E/NxpFwDnld( 1858): free library SUCCESS !!
E/NxpFwDnld( 1858): Freeing Mem for Dnld Context..
E/NxpHal  ( 1858): FW Download failed - NFCC init will continue
D/NfcAdaptation( 1858): NfcAdaptation::HalDeviceContextCallback: event=6
I/BrcmNfcNfa( 1858): nfc_main_hal_cback event: HAL_NFC_ERROR_EVT(0x6), status=1
I/BrcmNfcNfa( 1858): nfc_main_handle_hal_evt(): HAL event=0x6

Do you know what causes this?

Regards,
James.

jgeslin, 2017-03-24:
pn54x_dev_open: 10,57: 10 and 57 relates to inode reference, this is kernel internal reference.

jgeslin, 2017-03-24:
> It is pn544 to test now. Is it irrelevant?
I'm confused: which NXP NFC Controller are you using ?
even if kernel driver reference pn544 (previous NXP NFC Controller generation) it is applicable to more recent NXP NFC Controllers, however the NCI stack is not supporting PN544.

Jameskody, 2017-03-24:
The controller used is PN544.

Jameskody, 2017-03-24:
Is it supported in the Android Lollipop version?

Jameskody, 2017-03-24:
If PN544 is not supported,
If so, can you tell me how to support PN544?

jgeslin, 2017-03-24:
PN544 is supported via AOSP libnfc-nxp library, not through libnfc-nci one since it is not an NCI based controller.

mirh, 2017-12-18:
Soo... What's up with the landing of [this stuff](https://source.codeaurora.org/quic/la/external/libnfc-nci/log/?h=libnfc-nci-project/master) over at CAF?

jgeslin, 2017-12-19:
This new [CAF ](https://source.codeaurora.org/quic/la/external/libnfc-nci/log/?h=libnfc-nci-project/master) repo is just a move of the original [nxpgonk ](https://github.com/nxpgonk/libnfc-nci ) repo.
This relates to NFC SW Stack for specific OS (KaiOS) and doesn't support PN544 neither.

As said before, PN544 is supported in Android AOSP using external/libnfc-nxp library (in place of libnfc-nci). You may have a look to google targets (see target makefiles) integrating PN544 (Google Nexus S and Nexus 7).

closed 2018-09-04 by jgeslin

mirh, 2018-09-07:
Just for the records though, [external/libnfc-nxp](https://android.googlesource.com/platform/external/libnfc-nxp/) is EOL since ages. 

If any, AFAIK [lineageOS repo](https://github.com/LineageOS/android_external_libnfc-nxp) is the whatever actual one. 

p.s. would you happen to know the meanings of model_id values? PN544 manual isn't public unfortunately..

jgeslin, 2018-09-10:
Only 2 model ID values should exists:
 '00' indicates stand alone PN544 NFC controller
 '10' indicates System-in-package PN65N (PN544 NFC Controller + secure element)

mirh, 2018-09-10:
In my searches though, I actually found at least 5 values to be possible: [0](https://groups.google.com/forum/#!msg/seek-for-android/JuRRFq0grdU/XNM_CHdMr0YJ), [1](https://gist.github.com/erikcas/ed56f24b64865818e776#file-gistfile1-txt-L1794), [10](https://forum.xda-developers.com/xperia-s/s-development/pure-aosp-lollipop-t2940839/page91#post57501058), [11](https://forum.xda-developers.com/sony-xperia-z1-compact/development/rom-t3657384/page46#post73967065), [12](https://redmine.replicant.us/issues/957). 
And oddly, (at least that) 00 seemed to also report the secure element.

TanujaGudaje, 2019-05-29:
Hello All,

I am integrating NFC(PN7150) to Android Oreo(8.1.0)
I am not able enable NFC from Android Settings, it seems blocked.

from Logs I am getting 

01-01 00:30:10.601  1897  2672 D NfcNciHal: crcChecksumVerifyIntegrity: filename=/data/vendor/nfc/nfaStorage.bin1
01-01 00:30:10.601  1897  2672 D NfcNciHal: crcChecksumVerifyIntegrity: filename=/data/vendor/nfc/nfaStorage.bin2
01-01 00:30:10.601  1897  2672 D NfcNciHal: crcChecksumVerifyIntegrity: filename=/data/vendor/nfc/nfaStorage.bin3
01-01 00:30:10.601  1897  2672 D NfcNciHal: crcChecksumVerifyIntegrity: filename=/data/vendor/nfc/nfaStorage.bin4
01-01 00:30:10.601  1897  2672 D NfcNciHal: crcChecksumVerifyIntegrity: filename=/data/vendor/nfc/nfaStorage.bin5
01-01 00:30:10.612  1897  2675 D NfcAdaptation: NfcAdaptation::NFCA_TASK: enter
01-01 00:30:10.612  1897  2675 I BrcmNfcNfa: GKI_run(): Start/Stop GKI_timer_update_registered!
01-01 00:30:10.613  1897  2676 D NfcAdaptation: NfcAdaptation::Thread: enter
01-01 00:30:10.617  1897  2677 I BrcmNfcNfa: NFC_TASK started.
01-01 00:30:10.618  1897  2672 D NfcAdaptation: NfcAdaptation::InitializeHalDeviceContext: enter
01-01 00:30:10.618  1897  2676 D NfcAdaptation: NfcAdaptation::Thread: exit
01-01 00:30:10.618  1897  2672 I NfcAdaptation: NfcAdaptation::InitializeHalDeviceContext: INfc::getService()
01-01 00:30:10.618  1897  2676 E BrcmNfcNfa: gki_task task_id=0 terminating
01-01 00:30:10.619   581   581 E HWComposer: getLayerReleaseFence: Invalid display
01-01 00:30:10.619   581   581 I chatty  : uid=1000(system) /system/bin/surfaceflinger identical 1 line
01-01 00:30:10.619   581   581 E HWComposer: getLayerReleaseFence: Invalid display
01-01 00:30:10.619   549   728 E SDM     : HWCDisplayNull::Present: Present !!!

NOTE : INfc service is not returning back.
Please suggest how to resolve this issue.

Kind Regards,
Tanuja Gudaje