NXPNFCLinux/nxp-pn5xx #10: Driver error while loading using insmod command

prattom, 2017-03-10:
I have a custom board based on imx6ul evk board and I was able to compile the driver successfully by following this guide https://community.nxp.com/docs/DOC-329733. But after booting OS when I try to load the driver using insmod command I have received the following error :--

_pn5xx_i2c: disagrees about version of symbol dev_warn
pn5xx_i2c: Unknown symbol dev_warn (err -22)
pn5xx_i2c: disagrees about version of symbol dev_err
pn5xx_i2c: Unknown symbol dev_err (err -22)
insmod: ERROR: could not insert module pn5xx_i2c.ko: Invalid parameters_

I am using Yocto and my kernel version is Linux imx6ulevk 4.1.15-1.1.1+gd5d7c02.

jgeslin, 2017-03-17:
It seems that there is misalignment between "dev_warn" & "dev_err" functions called from the driver and the ones provided in your kernel version.
First check these definitions in your kernel: include/linux/device.h and verify it matches how there are used in the pn5xx_i2c driver. Eventually update pn5xx_i2c driver implementation to match the functions parameters.

prattom, 2017-03-18:
It was kernel definition problem.

closed 2017-03-18 by prattom