NXPNFCLinux/nxp-pn5xx #12: NFC hang

wintonliuwen, 2017-04-12:
Using PN7120
On Android,we touched NFC card 1000+ times continually and the NFC doesn't work.
There is no log of NxpNci and there is no interrupt from driver.
If I switch NFC off->on, and NFC works again.
That's the FW Version.
12-31 20:18:46.689  1033  1228 D NxpHal  : NxpNci> FW Version: 8.2.2

Is there a tool to debug whether the NFC module is working ?
If there are any other information to analysis this problem?

jgeslin, 2017-04-12:
Hi,

Do you mean NFC stopped working after 1000+ taps of the card ?
If so, first thing would be to check the ADB logs at the time the NFC stopped working. 
It may be that the PN7120 is stucked generating unmodulated field or even not generating field anymore. This could be come from different sources (for instance Android library may missed the card discovery notification and didn't restarted the discovery loop). The ADB logs should help identifying what's going on there.

Regards,
Jeremy.

wintonliuwen, 2017-04-13:
Hi Jeremy,

Yes, NFC works well before, after 1000+ taps, NFC stop work.
That the [log](https://www.dropbox.com/sh/odweiza0wh05keg/AADT58ILWTEcpoRI2i2cZSdAa?dl=0).
I didn't see any abnormal logs. Could you help to anlysis that?
Thank you.

Best regards,
Winton

jgeslin, 2017-04-13:
Hi,

Supposing aplog.1 file relates to the logs at the point NFC stopped working, we can observe the following trace:
04-03 22:34:13.949  1064  1265 D NxpNciR : len = 258 > 01FFFF0000121071616A4C6A5848785A5357396E79533800FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF

This is of course not a valid NCI answers, (first 3 bytes are erroneous, leading to wrong interpretation of NCI response). From my mind this comes from the I2C driver, if you can't solve it I would suggest to implement a recovery mechanism (triggering restart of the nfcservice).

Regards,
Jeremy.

wintonliuwen, 2017-04-14:
Hi Jeremy,

Really appreciate for your help.
PN7120 <-->I2C<-->HOST
Do you think this erroneous data is from PN7120 or the I2C driver modified the data?
I2C is just a communication channel, if the I2C driver has problem, there will always be a problem.
Exclude first 3bytes 01FFFF,   0000121071616A4C6A5848785A5357396E79533800 this is a valid NCI answer, right ? Is it possible to search the PN7120 firmware source code whether it would send the data 01FFFF? 
Could you reproduce this issue when do 1000+ taps?

For quick  solution as your suggestion, could you tell us how to change the code to do a recovery mechanism ? We are not familiar with the NFC HAL code.

Best regards,
Winton

chunguang0554, 2017-04-24:
Dear  jgeslin, 

Thanks for your reply and kind help, it works after modifying code to skip first 3 bytes garbage head "01FFFF".  

We also found another NFC stop working problem, the same issue as Wintonliuwen faced, there is no response when touching NFC card, and also no interrupt generated from NFC module. It requires to turn off/on NFC button to let it work again.  

Here is the Android log when issue happened.

Thanks again and waiting for your reply!
============================================
01-02 06:12:12.596  1090  1305 D NxpNciR : len =   4 > 41030100
01-02 06:12:12.596  1090  1305 D NxpTml  : PN54X - Posting read message.....
01-02 06:12:12.596  1090  1308 D NxpHal  : read successful status = 0x0
01-02 06:12:12.597  1090  1288 D NxpHal  : Response timer stopped
01-02 06:12:12.597  1090  1288 D NxpHal  : Checking response
01-02 06:12:12.597  1090  1288 E NxpHal  : Sending last command for Recovery 
01-02 06:12:12.597  1090  1288 E NxpHal  : Invoking data callback!!
01-02 06:12:12.597  1090  1288 E NxpHal  : Out phNxpNciHal_core_initialized rsp_params:0
01-02 06:12:12.597  1090  1305 D NxpTml  : PN54X - Read requested.....
01-02 06:12:12.597  1090  1305 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 06:12:12.608  1090  1305 E NxpTml  : _i2c_read() [hdr] errno : 79
01-02 06:12:12.608  1090  1305 E NxpTml  : PN54X - Error in I2C Read.....
01-02 06:12:12.608  1090  1305 D NxpTml  : PN54X - Read requested.....
01-02 06:12:12.608  1090  1305 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 06:12:12.659  1090  1305 D NxpTml  : PN54X - I2C Read successful.....
01-02 06:12:12.659  1090  1305 D NxpNciR : len =  24 > 61051501010606FF010A00004ED9882B000104E006808000
01-02 06:12:12.659  1090  1305 D NxpTml  : PN54X - Posting read message.....
01-02 06:12:12.659  1090  1308 D NxpHal  : read successful status = 0x0
01-02 06:12:12.659  1090  1308 D NxpHal  : NxpNci: RF Interface = Frame RF
01-02 06:archermind@archermind:/work$ tail -n 200 nxp-log.txt 
01-02 06:12:12.399  1090  1305 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 06:12:12.402  1090  1306 D NxpNciX : len =  20 > 20021104A05E0101A0400101A0410104A0430100
01-02 06:12:12.402  1090  1306 D NxpTml  : PN54X - I2C Write successful.....
01-02 06:12:12.402  1090  1306 D NxpTml  : PN54X - Posting Fresh Write message.....
01-02 06:12:12.402  1090  1306 D NxpTml  : PN54X - Tml Writer Thread Running................
01-02 06:12:12.402  1090  1308 D NxpHal  : write successful status = 0x0
01-02 06:12:12.402  1090  1288 D NxpHal  : Response timer started
01-02 06:12:12.402  1090  1288 D NxpHal  : Waiting after ext cmd sent
01-02 06:12:12.421  1090  1305 D NxpTml  : PN54X - I2C Read successful.....
01-02 06:12:12.422  1090  1305 D NxpNciR : len =   5 > 4002020000
01-02 06:12:12.422  1090  1305 D NxpTml  : PN54X - Posting read message.....
01-02 06:12:12.422  1090  1308 D NxpHal  : read successful status = 0x0
01-02 06:12:12.422  1090  1308 D NxpHal  : phNxpNciHal_print_res_status: response status =STATUS_OK
01-02 06:12:12.422  1090  1305 D NxpTml  : PN54X - Read requested.....
01-02 06:12:12.422  1090  1305 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 06:12:12.422  1090  1288 D NxpHal  : Response timer stopped
01-02 06:12:12.422  1090  1288 D NxpHal  : Checking response
01-02 06:12:12.422  1090  1288 D NxpExtns: const nxp::CNfcParam* nxp::CNfcConfig::find(const char*) const found NXP_CORE_RF_FIELD= �b
01-02 06:12:12.422  1090  1306 D NxpTml  : PN54X - Write requested.....
01-02 06:12:12.423  1090  1306 D NxpTml  : PN54X - Invoking I2C Write.....
01-02 06:12:12.424  1090  1306 D NxpNciX : len =   8 > 20020501A0620101
01-02 06:12:12.424  1090  1306 D NxpTml  : PN54X - I2C Write successful.....
01-02 06:12:12.424  1090  1306 D NxpTml  : PN54X - Posting Fresh Write message.....
01-02 06:12:12.425  1090  1306 D NxpTml  : PN54X - Tml Writer Thread Running................
01-02 06:12:12.426  1090  1308 D NxpHal  : write successful status = 0x0
01-02 06:12:12.426  1090  1288 D NxpHal  : Response timer started
01-02 06:12:12.426  1090  1288 D NxpHal  : Waiting after ext cmd sent
01-02 06:12:12.434  1090  1305 E NxpTml  : _i2c_read() [hdr] errno : 79
01-02 06:12:12.434  1090  1305 E NxpTml  : PN54X - Error in I2C Read.....
01-02 06:12:12.434  1090  1305 D NxpTml  : PN54X - Read requested.....
01-02 06:12:12.435  1090  1305 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 06:12:12.445  1090  1305 D NxpTml  : PN54X - I2C Read successful.....
01-02 06:12:12.445  1090  1305 D NxpNciR : len =   5 > 4002020000
01-02 06:12:12.445  1090  1305 D NxpTml  : PN54X - Posting read message.....
01-02 06:12:12.445  1090  1308 D NxpHal  : read successful status = 0x0
01-02 06:12:12.445  1090  1308 D NxpHal  : phNxpNciHal_print_res_status: response status =STATUS_OK
01-02 06:12:12.446  1090  1288 D NxpHal  : Response timer stopped
01-02 06:12:12.446  1090  1305 D NxpTml  : PN54X - Read requested.....
01-02 06:12:12.446  1090  1288 D NxpHal  : Checking response
01-02 06:12:12.446  1090  1305 D NxpTml  : PN54X - Invoking I2C Read.....
[... 134 lines cut ...]
01-02 06:12:12.588  1090  1306 D NxpTml  : PN54X - Posting Fresh Write message.....
01-02 06:12:12.589  1090  1306 D NxpTml  : PN54X - Tml Writer Thread Running................
01-02 06:12:12.589  1090  1308 D NxpHal  : write successful status = 0x0
01-02 06:12:12.589  1090  1288 D NxpHal  : Response timer started
01-02 06:12:12.590  1090  1288 D NxpHal  : Waiting after ext cmd sent
[... cut ...]
01-02 06:12:12.608  1090  1305 D NxpTml  : PN54X - Read requested.....
01-02 06:12:12.608  1090  1305 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 06:12:12.659  1090  1305 D NxpTml  : PN54X - I2C Read successful.....
01-02 06:12:12.659  1090  1305 D NxpNciR : len =  24 > 61051501010606FF010A00004ED9882B000104E006808000
01-02 06:12:12.659  1090  1305 D NxpTml  : PN54X - Posting read message.....
01-02 06:12:12.659  1090  1308 D NxpHal  : read successful status = 0x0
01-02 06:12:12.659  1090  1308 D NxpHal  : NxpNci: RF Interface = Frame RF

jgeslin, 2017-04-24:
Hi,

The last NCI exchange (reception of "61051501010606FF010A00004ED9882B000104E006808000" notification from the PN7120) shows that ISO15693 tag has been discovered and activated. Then there is no more exchange while we should expect the NFC service to, at least check if tag contains NDEF message and afterwards do presence check procedure to detect when tag is removed. The NFC stack looks to be locked at some point of the tag discovery notification raises across the differents layers. Unfortunately, since you filtered "Nxp" only traces we cannot foresee where it is locked.

Regards,
Jeremy.

chunguang0554, 2017-04-25:
Thanks for your quick reply, Jgeslin.
We will try again to see if it still can be reproduced, and then capture the whole log info.

chunguang0554, 2017-04-28:
Hello Jgeslin,

Issue reproduced, please refer to attached log file, thanks~

[aplog.zip](https://github.com/NXPNFCLinux/nxp-pn5xx/files/963272/aplog.zip)

chunguang0554, 2017-04-28:
Got reproduced again, please refer to log file attached, thanks~

[aplog-2.zip](https://github.com/NXPNFCLinux/nxp-pn5xx/files/963847/aplog-2.zip)

jgeslin, 2017-05-03:
Hi,

Looking to the first log (aplog.zip) I see some discrepancies in the I2C driver: 
> 12-31 19:32:37.532  1063  1253 D NxpNciR : len =   4 > 11010141

the received data is not an NCI frame. I'm not sure this is the cause of the deadlock because the NFC service restarts anyhow. However, the log ends at a point the MIFARE DESFire is being discovered and notified by PN7120:

> 12-31 19:32:40.319  1063  1253 D NxpNciR : len =  32 > 61051D01020400FF010C440307044A568A843C80012000000006057577810280

but the NFC service doesn't handle this information (no more logs from "1063" pid) !

Regarding the second log (aplog-2.zip), the behavior is totally different. Here we see that after several activation and communication with successively a MIFARE Classic card and a P2P device, the PN7120 raise a strange discovery notification (never observed before) of both type B and Type F in poll mode. This leads the NFC service to restart the discovery but only with type F poll mode enabled (and all listen mode). Afterwards only a type F listen device (P2P device supporting passive type F mode or Type 3 tag) can be discovered until the NFC service is restarted. I looking now to see from where comes this unexpected behavior of the NFC service. Could you please indicates which OS version are you running ?

Regards,
Jeremy.

chunguang0554, 2017-05-03:
Hello  Jgeslin,
The OS version of DUT is android 6.0.1

chunguang0554, 2017-05-09:
Any new finding ?

jgeslin, 2017-05-09:
Issue cannot be reproduced in my side. 
Please enable more traces, updating libnfc-brcm.conf file with:
`APPL_TRACE_LEVEL=0xFF`
`PROTOCOL_TRACE_LEVEL=0xFF`

chunguang0554, 2017-05-15:
Thanks geslin, i will try as your advice.

chunguang0554, 2017-06-12:
Hello Geslin, 
Turn on NFC trace switch, and got reproduced this issue again, please refer to the log file attached, thanks~
[nfc-log-0612.txt](https://github.com/NXPNFCLinux/nxp-pn5xx/files/1067269/nfc-log-0612.txt)

chunguang0554, 2017-06-12:
01-01 04:34:04.785  1056  1253 D NxpNciX : len =  26 > 2103170B00010101020103010501800182018301850106017701
01-01 04:34:04.785  1056  1253 D NxpTml  : PN54X - I2C Write successful.....
01-01 04:34:04.785  1056  1253 D NxpTml  : PN54X - Posting Fresh Write message.....
01-01 04:34:04.786  1056  1253 D NxpTml  : PN54X - Tml Writer Thread Running................
01-01 04:34:04.786  1056  1255 D NxpHal  : write successful status = 0x0
01-01 04:34:04.786  1056  1244 D NxpHal  : Response timer started
01-01 04:34:04.786  1056  1244 D NxpHal  : Waiting after ext cmd sent
01-01 04:34:04.791  1056  1252 E NxpTml  : _i2c_read() [hdr] errno : 79
01-01 04:34:04.791  1056  1252 E NxpTml  : PN54X - Error in I2C Read.....
01-01 04:34:04.791  1056  1252 D NxpTml  : PN54X - Read requested.....
01-01 04:34:04.791  1056  1252 D NxpTml  : PN54X - Invoking I2C Read.....
01-01 04:34:04.794  1056  1252 D NxpTml  : PN54X - I2C Read successful.....
01-01 04:34:04.794  1056  1252 D NxpNciR : len =   4 > 41030100
01-01 04:34:04.794  1056  1252 D NxpTml  : PN54X - Posting read message.....
01-01 04:34:04.794  1056  1255 D NxpHal  : read successful status = 0x0
01-01 04:34:04.795  1056  1252 D NxpTml  : PN54X - Read requested.....
01-01 04:34:04.795  1056  1244 D NxpHal  : Response timer stopped
01-01 04:34:04.795  1056  1252 D NxpTml  : PN54X - Invoking I2C Read.....
01-01 04:34:04.795  1056  1244 D NxpHal  : Checking response
01-01 04:34:04.795  1056  1244 E NxpHal  : Sending last command for Recovery 
01-01 04:34:04.795  1056  1244 E NxpHal  : Invoking data callback!!
01-01 04:34:04.795  1056  1244 D NfcAdaptation: NfcAdaptation::HalDeviceContextDataCallback: len=4
01-01 04:34:04.795  1056  1244 E NxpHal  : Out phNxpNciHal_core_initialized rsp_params:0
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: NFC received ntf gid:2
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nci_proc_ee_management_ntf opcode:0x0
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfcee_id: 0x2 num_interface:0x1/0x1, num_tlvs:0x0
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_dm_nfc_response_cback () NFC_NFCEE_INFO_REVT(0x5005)
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_ee_proc_evt: event=0x5005 int_event:0x20e
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_ee_evt_hdlr (): Event NCI_DISC_NTF(0x20e), State: INIT_DONE(1)
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: recived dis_ntf; stopping timer
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_sys_ptim_stop_timer 6ba7e368
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: ptim timer stop
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_ee_nci_disc_ntf() em_state:1 ee_flags:0x0 cur_ee:2 num_ee_expecting:0
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_ee_find_ecb ()
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: cur_ee:2 ecb_flags=0x00  ee_status=0x2
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_ee_nci_disc_ntf cur_ee:2
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: nfa_ee_report_disc_done() em_state:1 num_ee_expecting:0 notify_enable_done:0
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: NFA_AllEeGetInfo em_state:1 cur_ee:2
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: xx:0 max_ret:3, num_ret:0 ee_status:0x0
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: xx:1 max_ret:3, num_ret:1 ee_status:0x2
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: num_ret:2
01-01 04:34:04.795  1056  1244 D BrcmNfcJni: nfaEeCallback: NFA_EE_DISCOVER_EVT; status=0x0; num ee=2
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: NFC received rsp gid:1
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: resetting the nci_cmd_window
01-01 04:34:04.795  1056  1244 I BrcmNfcNfa: p2p priority is running under bail out mode ONLY.
01-01 04:34:04.796  1056  1244 I BrcmNfcNfa: nfa_dm_disc_discovery_cback (): event:0x4000
01-01 04:34:04.796  1056  1244 I BrcmNfcNfa: nfa_dm_disc_sm_execute (): state: DISCOVERY (1), event: DISCOVER_RSP(1) disc_flags: 0x21
01-01 04:34:04.796  1056  1244 E BrcmNfcNfa: nfa_dm_disc_sm_discovery (): Unexpected discovery event
01-01 04:34:04.796  1056  1244 I BrcmNfcNfa: nfa_dm_disc_sm_execute (): new state: DISCOVERY (1), disc_flags: 0x21
01-01 04:34:04.796  1056  1244 E BrcmNfcNfa: nfc_ncif_update_window: Unexpected call
01-01 04:34:04.801  1056  1252 E NxpTml  : _i2c_read() [hdr] errno : 79
01-01 04:34:04.801  1056  1252 E NxpTml  : PN54X - Error in I2C Read.....
01-01 04:34:04.801  1056  1252 D NxpTml  : PN54X - Read requested.....
01-01 04:34:04.801  1056  1252 D NxpTml  : PN54X - Invoking I2C Read.....
01-01 04:34:04.824  1056  1252 D NxpTml  : PN54X - I2C Read successful.....
01-01 04:34:04.824  1056  1252 D NxpNciR : len =  32 > 61051D01020400FF010C44030704095512604380012000000006057577810280
01-01 04:34:04.824  1056  1252 D NxpTml  : PN54X - Posting read message.....
01-01 04:34:04.825  1056  1255 D NxpHal  : read successful status = 0x0
01-01 04:34:04.825  1056  1255 D NxpHal  : NxpNci: RF Interface = ISO-DEP
01-01 04:34:04.825  1056  1255 D NxpHal  : NxpNci: Protocol = ISO-DEP
01-01 04:34:04.825  1056  1255 D NxpHal  : NxpNci: Mode = A Passive Poll
01-01 04:34:04.825  1056  1255 D NfcAdaptation: NfcAdaptation::HalDeviceContextDataCallback: len=32
01-01 04:34:04.825  1056  1252 D NxpTml  : PN54X - Read requested.....
01-01 04:34:04.825  1056  1252 D NxpTml  : PN54X - Invoking I2C Read.....
01-01 04:34:04.825  1056  1244 I BrcmNfcNfa: NFC received ntf gid:1
01-01 04:34:04.825  1056  1244 I BrcmNfcNfa: p2p priority is running under bail out mode ONLY.
01-01 04:34:04.825  1056  1244 I BrcmNfcNfa: nfc_set_state 5 (OPEN)->5 (OPEN)
[... cut ...]
01-01 04:34:04.825  1056  1244 I BrcmNfcNfa: nfa_dm_disc_sm_execute (): state: DISCOVERY (1), event: ACTIVATED_NTF(5) disc_flags: 0x21
01-01 04:34:04.825  1056  1244 I BrcmNfcNfa: RF Activated while waiting for deactivation RSP
01-01 04:34:04.825  1056  1244 I BrcmNfcNfa: nfa_dm_disc_sm_execute (): new state: DISCOVERY (1), disc_flags: 0x61
01-01 04:34:04.832  1056  1252 E NxpTml  : _i2c_read() [hdr] errno : 79
01-01 04:34:04.832  1056  1252 E NxpTml  : PN54X - Error in I2C Read.....
01-01 04:34:04.832  1056  1252 D NxpTml  : PN54X - Read requested.....
01-01 04:34:04.832  1056  1252 D NxpTml  : PN54X - Invoking I2C Read.....

chunguang0554, 2017-06-14:
Hello Geslin, Could you please analyze log, and find the reason why NFC stops working ?

chunguang0554, 2017-06-21:
Hello Geslin,
 this issue is critical to us, could you help take a look and analyze the reason, many thanks to you!

jgeslin, 2017-06-21:
Right now I can only advise you to kill the NFC service in such recovery condition.
In nfc_ncif_cmd_timeout() function (inside nfc_ncif.c file), insert "exit(1);" instruction:

> void nfc_ncif_cmd_timeout (void)
{
    NFC_TRACE_ERROR0 ("nfc_ncif_cmd_timeout");

> #if(NXP_EXTNS == TRUE)

>    NFC_TRACE_ERROR0 ("Recovery Start!");
    exit(1);
    {
        // TO DO:
        // Sending command which are core_reset,init,discovery as raw packet (Hard coding)
...

chunguang0554, 2017-06-23:
Appreciate your help again, Jgeslin！
I will follow your advice, and try again.

chunguang0554, 2017-07-05:
Hello Jgeslin,

NFC stop working again after running nearly 3 hours and a half. This time it looks like another reason, and the last log message has been posted here, and you could click here(
[nfc_log_0705.txt](https://github.com/NXPNFCLinux/nxp-pn5xx/files/1123212/nfc_log_0705.txt)
) to get full logs  

01-02 03:22:23.913  1033  1219 D NxpTml  : PN54X - I2C Read successful.....
01-02 03:22:23.913  1033  1219 D NxpNciR : len =   4 > 41060100
01-02 03:22:23.913  1033  1219 D NxpTml  : PN54X - Posting read message.....
01-02 03:22:23.914  1033  1222 D NxpHal  : read successful status = 0x0
01-02 03:22:23.914  1033  1219 D NxpTml  : PN54X - Read requested.....
01-02 03:22:23.914  1033  1219 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 03:22:23.917   307   646 D AUD_HAL : afe_hal_setup_afe: Sample rate for AFE_IN path = 3
01-02 03:22:23.919   307   646 D AUD_HAL : afe_hal_setup_afe: Sample rate for AFE_IN path = 3
01-02 03:22:23.921   307   646 W AUD_HAL : dsp_aud_power_save_disable: Enabling Audio DSP power
01-02 03:22:23.924   307   646 D AUD_HAL : aud_set_control:Trying to set control DSP Audio Power to value 1 
01-02 03:22:23.925   307   646 D AUD_HAL : afe_hal_setup_afe: Sample rate for AFE_IN path = 3
01-02 03:22:23.927   307   659 D AUD_HAL : Active source UTA_AUDIO_SOURCE_PCM_GENERAL
01-02 03:22:23.927   307   659 W AUD_HAL : mode = AUDIO_HW_ILLEGAL_MODE for source UTA_AUDIO_SOURCE_PCM_GENERAL
01-02 03:22:23.927   307   659 D AUD_HAL : Connected route from UTA_AUDIO_SOURCE_PCM_GENERAL to UTA_AUDIO_DESTINATION_ANALOG_OUT
01-02 03:22:23.927   307   659 D AUD_HAL : UTA_AUDIO_DESTINATION_ANALOG_OUT destination configured to UTA_AUDIO_ANALOG_DEST_HEADSET_PLUS_BACKSPEAKER
01-02 03:22:23.927   307   659 D AUD_HAL : Mails are pending in audio driver event queue -> sleep 20 ms
01-02 03:22:23.927   307   646 D AUD_HAL : afe_hal_setup_afe: set left analog gain 32 , for analog_gain = 6
01-02 03:22:23.927   307   646 D AUD_HAL : afe_hal_setup_afe: set right analog gain 10 , for analog_gain = 6
01-02 03:22:23.927   307   646 D AUD_HAL : afe_hal_setup_afe: Sample rate for AFE_IN path = 3
01-02 03:22:23.927  1033  1219 D NxpTml  : PN54X - I2C Read successful.....
01-02 03:22:23.927  1033  1219 D NxpNciR : len =   5 > 6106026106
01-02 03:22:23.928  1033  1219 D NxpTml  : PN54X - Posting read message.....
01-02 03:22:23.928  1033  1222 D NxpHal  : read successful status = 0x0
01-02 03:22:23.928  1033  1219 D NxpTml  : PN54X - Read requested.....
01-02 03:22:23.928  1033  1219 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 03:22:23.928   307   646 D AUD_HAL : aud_set_control:Trying to set control LS ClassD Frequency Divider to value 35 
01-02 03:22:23.928  1033  1213 E BrcmNfcJni: getReconnectState = 0x0
01-02 03:22:23.928  1033  1213 D BrcmNfcJni: is_ee_recovery_ongoing : recovery
01-02 03:22:23.928  1033  1033 E BrcmNfcJni: setReconnectState = 0x0
01-02 03:22:23.929  1033  1033 E BrcmNfcJni: nativeNfcTag_doCheckNdef: tag already deactivated
01-02 03:22:23.929  1033  1033 D NativeNfcTag: Check NDEF Failed - status = 3
01-02 03:22:23.929  1033  1033 D NativeNfcTag: connectWithStatus- technology =9
01-02 03:22:23.929  1033  1033 D NativeNfcTag: connectWithStatus- technology =9
01-02 03:22:23.929  1033  1033 E BrcmNfcJni: nativeNfcTag_doConnect: tag already deactivated
01-02 03:22:23.929  1033  1033 D NativeNfcTag: Connect Failed - status = 255
01-02 03:22:23.929   307   646 D AUD_HAL : aud_set_control:Trying to set control AFE CP Frequency Control to value 13000 
01-02 03:22:23.929  1033  1033 E BrcmNfcJni: nativeNfcTag_doReconnect: tag already deactivated
01-02 03:22:23.929  1033  1033 E BrcmNfcJni: nativeNfcTag_doDisconnect: tag already deactivated
01-02 03:22:23.932  1033  1257 W AudioTrack: AUDIO_OUTPUT_FLAG_FAST denied by client; transfer 4, track 16000 Hz, output 48000 Hz
01-02 03:22:23.937   307   646 D AUD_HAL :  AUD_ACTIVATE AFE 1
01-02 03:22:23.937   307   646 W AUD_HAL : aud_activate_afe: Opening IMAS pcm devices
01-02 03:22:23.938   307   646 W AUD_HAL : aud_activate_afe: Starting IMAS pcm devices
01-02 03:22:23.939  1033  1219 E NxpTml  : _i2c_read() [hdr] errno : 79
01-02 03:22:23.939  1033  1219 E NxpTml  : PN54X - Error in I2C Read.....
01-02 03:22:23.939  1033  1219 D NxpTml  : PN54X - Read requested.....
01-02 03:22:23.939  1033  1219 D NxpTml  : PN54X - Invoking I2C Read.....
01-02 03:22:23.940   307   646 W AUD_HAL : AUDIO_DSP_CMD_VB_HW_AFE:  0003 0003

jgeslin, 2017-07-06:
Hi,

Indeed the behavior is slightly different, even if the same root cause: wrong data received from the low I2C communication level (01-02 03:22:23.277) causing same issue: misalignement of state machine.
I would advise you to perform a recovery (as previously proposed) in the case of an invalid NCI frame received: basically received frame must always start with 0x4X or 0x6X or 0x00. This should already limit the issue appearence. Then you could detect misalignement and perform the recovery as well (e.g. "nativeNfcTag_doDisconnect: tag already deactivated").

Regards,
Jeremy.

chunguang0554, 2017-07-10:
Appreciate your help, Jgeslin, and i will add code to check received frame, and drop those invalid frame.

Another question i would like to consult you.
Our NFC chipset is PN7120, which supports bitrates listed below:
106 kbps (Type A,B)    
212 kbps (Type A,B& F)    
424 kbps (Type A,B & F)     
848 kbps (Type A,B)

If i wants to disable 848kbps and 424kbps, do you know how to implements that ?

I searched in the code, and found one fouction "static NFCSTATUS phNxpNciHal_uicc_baud_rate()" has the similiar feature, but it only suit for PN547C2 chipset. 

It defined getUiccBitRateBuff array, and setUiccBitRateBuf array, which should be the core package format to set/get chipset bitrate.   i think if i followed the same way to modify code, it should reach to the target.  But i don't have NCI spec, and don't know what's the right package content for PN7120, could you do me a favor to give me some advice ?
    uint8_t  setUiccBitRateBuf[] = {0x20, 0x02, 0x01, 0x00, 0xA0, 0x86, 0x01, 0x91, 0xA0, 0x87, 0x01, 0x91};
    uint8_t  getUiccBitRateBuf[] = {0x20, 0x03, 0x05, 0x02, 0xA0 ,0x86 ,0xA0 , 0x87};

Thanks again!

jgeslin, 2017-08-01:
Hi,

Sorry for the late response.
The UiccBitRate parameters relates to datarate used on SWP, which is not supported by PN7120.
To prevent NFC Controller using higher baudrate, several NCI standard parameters must be properly set: 
- PI_BIT_RATE = 0x0 (to restrict ISO14443 communication to 106kbits/s)
- PF_BIT_RATE = 0x1 (to restrict Type F communication to 212kbits/s)
- PN_NFC_DEP_SPEED = 0x1 (to prevent switching to higher bitrate after NFC-DEP activation)

Setting those parameters in Android environment is done via NXP_CORE_CONF flag defined in libnfc-nxp.conf file. See below in bold the related parameters (based on Lollipop release):

NXP_CORE_CONF={ 20, 02, 2B, 0D,
        **28, 01, 01,              // NFC_DEP_SPEED set not to switch to higher bitrate**
        **21, 01, 00,              // PI_BIT_RATE set to 106 kbits/s**
        30, 01, 08,
        31, 01, 03,
        33, 04, 01, 02, 03, 04,
        54, 01, 06,
        50, 01, 02,
        5B, 01, 00,
        60, 01, 0E,
        80, 01, 01,
        81, 01, 01,
        82, 01, 0E,
        **18, 01, 01            // PF_BIT_RATE set to 212 kbits/s**
        }

For your information, the syntax of this field is the CORE_SET_CONFIG NCI standard command (defined in NFC Forum NCI specification) where first bytes "20, 02" relates to the command header, the following "2B" the remaining payload size and "0D" the number of parameters contained in the command. Then for each parameters, the first bytes is the parameter ID (see NCI spec), the second one the parameter value length and last bytes are the parameter value to set.

Regards,

Jeremy.

closed 2018-09-04 by jgeslin