NXPNFCLinux/nxp-pn5xx #16: Installing driver on a Raspberry Pi w/o compiling a custom kernel

kroegerama, 2017-06-02:
Is there any tutorial on how to install the pn5xx driver on a clean Debian installation?
I'm trying for hours now without any luck. There is hardly any information on the internet, and the NXP App Notes are useless for this case.

The demo image works fine, but I want to setup a less bloated system.

Are a compiled .ko module and a .dtb file available to download anywhere?

jgeslin, 2017-06-06:
Indeed, the pn5xx_i2c driver can be compiled as a single module, (see generic instruction here: https://debian-administration.org/article/640/Rebuilding_a_single_kernel_module). So you need at least the headers files related to your kernel version.
The DTS must also be generated with the inclusion of the pn5xx declaration (from the original DTS of your kernel version).
This is basically described in chapter 3 of [AN11697](http://www.nxp.com/documents/application_note/AN11697.pdf).

In the meantime we are preparing a new release of Linux_libnfc-nci library offering the possibility to workaround the use of the pn5xx_i2c kernel module by directly accessing the platform I2C and GPIO drivers.
This should be made available [here](https://github.com/NXPNFCLinux/linux_libnfc-nci) in the coming weeks.

kroegerama, 2017-06-07:
Thanks for your reply! I will try to follow your linked instructions.

The new libnfc-nci library sounds very interesting. Is there any way to get (alpha/beta) preview access?

jgeslin, 2017-06-13:
The alternative to pn5xx_i2c kernel driver is now part of R2.R release of the [libnfc-nci](https://github.com/NXPNFCLinux/linux_libnfc-nci). The documentation related update will follow soon, however you can already use it: 
- at configure step, uses "--enable-alt" option 
- make sure proper definition is done in phTml_Nfc_alt.h (for raspberry pi + PN71xx demo kit set CONFIGURATION variable to 1)
- insure access rights to /sys/class/gpio and /dev/i2c-1 ressources are given to the application or run the application with root privileges

Regards,
Jeremy.

closed 2018-09-04 by jgeslin