NXPNFCLinux/nxp-pn5xx #20: Feature request: add ACPI support

jr64, 2018-08-10:
A lot of modern ThinkPad models (and also other notebooks) include an NXP chip (NPC300) that would work with this driver. Most desktop Linux distributions don't use device tree so support for ACPI is required to make the NFC devices work. It would be great if you could add it because it would make the built-in NFC work on a ton of notebooks.

I have already gotten started on implementing ACPI support (see my fork @ https://github.com/jr64/nxp-pn5xx). Currently, I don't think the code is mature enough to be included in the upstream repo but would you be interested in a pull request once it is?

My problems currently are:
* There are three gpio pins defined in the DSDT. I correctly identified ven and irq but I'm not sure what the other one is. I am assuming it is either clkreq or firmware? Is there any way to check? If there are only three pins, does that mean that the fourth is not connected?
* the device tree attempts to read and configure four regulator lines, any way to get this information through ACPI? I'm assuming not but just making sure.
* I'm not sure if this is directly related to the i2c driver but whenever I use the NFC device, the touchpad of my notebook becomes slow and unresponsive and the cursor sometimes jumps around. This even persists once I unload the kernel driver (until reboot). Any idea what could be causing this?

Any help would be greatly appreciated!

DSDT:
```
Scope (\_SB.PCI0)
    {
        Name (RID, 0x00)
        Scope (I2C0)
        {
            Device (NFC1)
            {
                Name (_ADR, 0x00)  // _ADR: Address
                Name (_HID, "NXP1001")  // _HID: Hardware ID
                Name (_DDN, "NXP NFC For Win10")  // _DDN: DOS Device Name
                Name (_UID, 0x01)  // _UID: Unique ID
                Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
                {
                    Name (RBUF, ResourceTemplate ()
                    {
                        I2cSerialBusV2 (0x0029, ControllerInitiated, 0x00061A80,
                            AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                        GpioInt (Level, ActiveHigh, Exclusive, PullNone, 0x0000,
                            "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x006C
                            }
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0027
                            }
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0064
                            }
                    })
                    Return (RBUF) /* \_SB_.PCI0.I2C0.NFC1._CRS.RBUF */
                }

                Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    If ((\OSYS >= 0x07DF))
                    {
                        If (((\_SB.GGIV (0x02030016) == 0x00) && (\NFCF == 0x00)))
                        {
                            Return (0x0F)
                        }
                    }

                    Return (0x00)
                }
            }
        }
    }
```

jr64, 2018-08-10:
I'm fairly certain now that the behavior of the touchpad is due to bugs in the Elan driver and has nothing to do with the NXP driver discussed here.

eqvinox, 2019-06-15:
For reference, here's the DSDT for an HP EliteBook 830 G5:

```
    Scope (_SB.PCI0.I2C1)
    {
        Device (NFC1)
        {
            Name (_ADR, Zero)  // _ADR: Address
            Name (_HID, EisaId ("NXP1002"))  // _HID: Hardware ID
            Name (_CID, "NXP1002")  // _CID: Compatible ID
            Name (_DDN, "NXP NFC")  // _DDN: DOS Device Name
            Name (_UID, One)  // _UID: Unique ID
            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                Name (SBUF, ResourceTemplate ()
                {
                    I2cSerialBusV2 (0x0029, ControllerInitiated, 0x00061A80,
                        AddressingMode7Bit, "\\_SB.PCI0.I2C1",
                        0x00, ResourceConsumer, _Y2C, Exclusive,
                        )
                    GpioInt (Edge, ActiveHigh, ExclusiveAndWake, PullDefault, 0x0000,
                        "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                        )
                        {   // Pin list
                            0x0000
                        }
                    GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                        "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                        )
                        {   // Pin list
                            0x0000
                        }
                    GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                        "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                        )
                        {   // Pin list
                            0x0000
                        }
                })
                CreateWordField (SBUF, \_SB.PCI0.I2C1.NFC1._CRS._Y2C._ADR, NADR)  // _ADR: Address
                CreateDWordField (SBUF, \_SB.PCI0.I2C1.NFC1._CRS._Y2C._SPE, NSPD)  // _SPE: Speed
                CreateWordField (SBUF, 0x38, NFCA)
                CreateWordField (SBUF, 0x60, NFCB)
                CreateWordField (SBUF, 0x88, NFCC)
                NADR = 0x29
                NSPD = 0x00061A80
                NFCA = GNUM (0x02010015)
                SHPO (0x02010015, One)
                NFCB = GNUM (0x0201000F)
                NFCC = GNUM (0x02010012)
                SHPO (0x0201000F, One)
                SHPO (0x02010012, One)
                Return (SBUF) /* \_SB_.PCI0.I2C1.NFC1._CRS.SBUF */
            }

            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                If ((NFCS == One))
                {
                    Return (0x0F)
                }

                Return (Zero)
            }
        }
    }
```

Also, I would strongly assume CLKREQ would be an output from the NFC controller and thus GPIO input — but the two non-IRQ IO pins here are outputs.