NXPNFCLinux/nxp-pn5xx #23: nxp-pn5xx driver freezes system on suspend/resume

jjvbsag, 2019-10-22:
We are currently implementing a new embedded device for industrial use.
Main features (relevant for this issue)

- ARM system based on [Digi ConnectCore® 6UL](https://www.digi.com/products/embedded-systems/system-on-modules/connectcore-for-i-mx6ul)
- linux-4.9 with rt-patches and CONFIG_HIBERNATION set
- NFC Tag reader using PN7150
- Powerfail safe using suspend to disk (industrial sd-card)

To operate the PN7150, we use the nxp-pn5xx driver and linux_libnfc-nci.
Now our problem is:
**when nxp-pn5xx driver is loaded, the whole system freezes, when we try to suspend to disk.**

- Is this a known problem? 
- Are there plans to solve this?
- Can you give hints, where and how to solve?
- Do you need further information?

We will happily verify any suggestion and also share a solution found.

jgeslin, 2019-10-24:
Hi,

Can you try to figure out which instruction (I2C management , GPIO management, other ...) , from the pn54x_probe() function, is preventing the suspend ?
You can comment out part of the code of this function, step-by-step, and check when the freeze is over.
This will help figuring out what is the root cause.

Regards,
Jeremy.

jjvbsag, 2019-10-25:
@jgeslin 
Ok, maybe I was a bit unclear. We have no issue with `pn54x_probe`. This is working, the system is booting and chip is functional. 

Our issue is: **There is no power management in the driver**. Especially, in the `pn54x_driver` the element `.pm` is not filled.

When we try to suspend to disk with `reboot(RB_SW_SUSPEND)`, the suspend is working, and also the resume is working until a specific step. But then we get messages
```
NOHZ: local_softirq_pending 102
NOHZ: local_softirq_pending 102
NOHZ: local_softirq_pending 102
NOHZ: local_softirq_pending 102
NOHZ: local_softirq_pending 102
```
and the resume gets stalled.

This is typically caused by a driver, where the chip is generating interrupts, but the system isn't aware anymore about the chip. This is clearly an issue of nxp-pn5xx, because if we don't load the driver, the resume completes without any problems.

Do you have any suggestions, how to implement PM in the driver? Typically there should be something like
```
static int pn54x_suspend(struct device *dev)
{
:::
}
static int pn54x_resume(struct device *dev)
{
:::
}
static SIMPLE_DEV_PM_OPS(pn54x_dev_pm_ops, pn54x_suspend, pn54x_resume);
:::
#ifdef CONFIG_PM
		.pm		= &pn54x_dev_pm_ops,
#endif /* CONFIG_PM */

```
where `pn54x_suspend` should disable all interrupt sources and release all interrupts and `pn54x_resume` should request the interrupts again and reinit the chip. 

As said, we will happily try this by ourself, but we are thankful for any suggestions, what to do in `pn54x_suspend` and `pn54x_resume`

jjvbsag, 2019-10-29:
Ok, with the following patch (which includes a lot of debug to be removed in the final version) we can suspend and resume without system freeze.

```diff
--- src/modules/nxp-pn5xx/pn5xx_i2c.c	(Revision 346706)
+++ src/modules/nxp-pn5xx/pn5xx_i2c.c	(Arbeitskopie)
@@ -1,3 +1,4 @@
+#define DEBUG
 /*
  * Copyright (C) 2010 Trusted Logic S.A.
  * modifications copyright (C) 2015 NXP B.V.
@@ -35,6 +36,7 @@
 #include <linux/gpio.h>
 #include <linux/miscdevice.h>
 #include <linux/spinlock.h>
+#include <linux/pm.h>
 #include "pn5xx_i2c.h"
 #include <linux/of_gpio.h>
 #include <linux/regulator/consumer.h>
@@ -69,7 +71,11 @@
 	struct regulator *pmuvcc_reg;
 	struct regulator *sevdd_reg;
 	bool irq_enabled;
+	int mode;
 	spinlock_t irq_enabled_lock;
+
+	bool pm_irq_enabled;
+	int pm_mode;
 };
 
 /**********************************************************
@@ -105,6 +111,8 @@
 static int pn544_enable(struct pn54x_dev *dev, int mode)
 {
 	int r;
+	pr_info("%s\n",__func__);
+	pr_info("> mode=%d\n",mode);
 
 	/* turn on the regulators */
 	/* -- if the regulators were specified, they're required */
@@ -171,7 +179,7 @@
 		pr_err("%s bad arg %d\n", __func__, mode);
 		return -EINVAL;
 	}
-
+	dev->mode=mode;
 	return 0;
 
 enable_exit2:
@@ -197,7 +205,7 @@
 	if(dev->pmuvcc_reg) regulator_disable(dev->pmuvcc_reg);
 	if(dev->vbat_reg) regulator_disable(dev->vbat_reg);
 	if(dev->pvdd_reg) regulator_disable(dev->pvdd_reg);
-
+	dev->mode=MODE_OFF;
 }
 
 /**********************************************************
@@ -723,6 +731,59 @@
 	return 0;
 }
 
+
+/*-------------------------------------------------------------------------*/
+#ifdef CONFIG_PM
+static int pn54x_resume(struct device *dev)
+{
+	int ret;
+	struct i2c_client *client = to_i2c_client(dev);
+	struct pn54x_dev *pn54x_dev = i2c_get_clientdata(client);
+	dev_dbg(dev,"%s\n",__func__);
+	dev_dbg(dev,"> client=%p\n",client);
+	dev_dbg(dev,"> pn54x_dev=%p\n",pn54x_dev);
+
+	if(pn54x_dev->pm_mode)
+		pn544_enable(pn54x_dev,pn54x_dev->pm_mode);
+	else
+		pn544_disable(pn54x_dev);
+
+	pn54x_dev->irq_enabled = true;
+	ret = request_irq(client->irq, pn54x_dev_irq_handler,
+				IRQF_TRIGGER_HIGH, client->name, pn54x_dev);
+	if (ret) {
+		dev_err(&client->dev, "request_irq failed\n");
+	}
+	pn54x_disable_irq(pn54x_dev);
+
+
+	return 0;
+}
+
+static int pn54x_suspend(struct device *dev)
+{
+	struct i2c_client *client = to_i2c_client(dev);
+	struct pn54x_dev *pn54x_dev = i2c_get_clientdata(client);
+	dev_dbg(dev,"%s\n",__func__);
+	dev_dbg(dev,"> client=%p\n",client);
+	dev_dbg(dev,"> pn54x_dev=%p\n",pn54x_dev);
+	pn54x_dev->pm_mode=pn54x_dev->mode;
+	pn54x_dev->pm_irq_enabled=pn54x_dev->irq_enabled;
+	dev_dbg(dev,"> pm_mode=%d\n",pn54x_dev->pm_mode);
+	dev_dbg(dev,"> pm_irq_enabled=%d\n",pn54x_dev->pm_irq_enabled);
+
+	pn54x_disable_irq(pn54x_dev);
+	pn544_disable(pn54x_dev);
+	free_irq(client->irq, pn54x_dev);
+
+	return 0;
+}
+
+static SIMPLE_DEV_PM_OPS(pn54x_dev_pm_ops, pn54x_suspend, pn54x_resume);
+
+#endif /* CONFIG_PM */
+
+
 /*
  *
  */
@@ -753,6 +814,9 @@
 		.owner	= THIS_MODULE,
 		.name	= "pn544",
 		.of_match_table = pn54x_dt_match,
+#ifdef CONFIG_PM
+		.pm		= &pn54x_dev_pm_ops,
+#endif /* CONFIG_PM */
 	},
 };
 

```

But after a resume the nfc is not functional any more. It seems, there is some initialization missing.
Following test:

- do a clean system boot
- start `nfcdemo poll` => NFC is working, tags are detected.
- do a suspend and a resume (without powercycle)

Then nfcdemo starts to generate messages and is not functional any more
```
54x_dev_write : i2c_master_send(28) returned -6
NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2Cpn54x_dev_write : i2c_master_send(28) returned -6
 Write.....

NxpHal:     write error status = 0x1ff
NxpHal:   pn54x_dev_write : i2c_master_send(28) returned -6
  write_unlocked failed - PN54X Maybe in Standby Mode - Retry
Nxpn54x_dev_write : i2c_master_send(28) returned -6
pTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2pn54x_dev_write : i2c_master_send(28) returned -6
C Write.....

NxpHal:     write error status = 0x1ff
NxpHal:  pn54x_dev_write : i2c_master_send(28) returned -6
   write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

```

Even killing and restarting nfcdemo does not change the messages.

jjvbsag, 2019-10-30:
Ok, another test.
After suspend and killing nfcdemo, doing a `modprobe -r pn5xx_i2c` and then  `modprobe pn5xx_i2c` returns the device into a functional state. But this is no real solution for us, because the application polling the nfc tag gets no notification of the problem (there is no error notification in the linux_nfc_api)

jjvbsag, 2019-11-04:
@jgeslin Maybe this requires additional fixes in [linux_libnfc-nci](https://github.com/NXPNFCLinux/linux_libnfc-nci) ? Shall I add an issue there?

jgeslin, 2019-11-04:
The issue is not linked to the linux_libnfc-nci. The suspend/resume should be totally hidden to the linux_libnfc-nci. To my mind i2c reported errors after `resume `relates to something "broken" during the `suspend`/`resume `sequence. You could maybe try to remove/add the pn54x driver from i2c definition during `suspend`/`resume `using `i2c_del_driver(&pn54x_driver)`/`i2c_add_driver(&pn54x_driver)`.

jjvbsag, 2019-11-04:
@jgeslin 
> The suspend/resume should be totally hidden to the linux_libnfc-nci.

That's also what I expected. 

Regarding your proposal: `i2c_del_driver(&pn54x_driver)` in `pn54x_suspend` causes a freeze during suspend.

Also I assume
- a driver cannot delete itself, while in a driver method.
- even if it could: Because it has been removed, `resume` would never be called.

jjvbsag, 2019-11-04:
Just took a look about other sources:
E.g. drivers/hwmon/lm75.c
```c

#ifdef CONFIG_PM
static int lm75_suspend(struct device *dev)
{
	int status;
	struct i2c_client *client = to_i2c_client(dev);
	status = i2c_smbus_read_byte_data(client, LM75_REG_CONF);
	if (status < 0) {
		dev_dbg(&client->dev, "Can't read config? %d\n", status);
		return status;
	}
	status = status | LM75_SHUTDOWN;
	i2c_smbus_write_byte_data(client, LM75_REG_CONF, status);
	return 0;
}

static int lm75_resume(struct device *dev)
{
	int status;
	struct i2c_client *client = to_i2c_client(dev);
	status = i2c_smbus_read_byte_data(client, LM75_REG_CONF);
	if (status < 0) {
		dev_dbg(&client->dev, "Can't read config? %d\n", status);
		return status;
	}
	status = status & ~LM75_SHUTDOWN;
	i2c_smbus_write_byte_data(client, LM75_REG_CONF, status);
	return 0;
}
```
They tell the chip not to generate any interrupts during suspend. 
Is this possible for PN7150 ?

jgeslin, 2019-11-04:
By calling `pn544_disable` during `suspend`, it prevents PN7150 raising IRQ line (so no interrupt will be generated).
However, I missed that in the changes you've done, `pn544_disable` / `pn544_enable` sequence is reseting PN7150 and the linux_libnfc-nci is not able to handle this reset (even if it would be notified). That may explain the behavior after resume. Can you try commenting out `pn544_disable` and `pn544_enable` from `suspend `and `resume `?

jjvbsag, 2019-11-05:
@jgeslin to reduce traffic :vertical_traffic_light: :car: :bus: :busstop: :car:  I would not post every try'n'fail :bomb: patch here :grin: 

But please note: we have to observer two different cases.
- **suspend/resume while supply stays stable** : 
This is the simple case, but even in this simple case I cannot get the PN7150 to communicate after resume. _Are you sure_, after doing `pn544_disable` and then `pn544_enable` PR7150 needs _no_ initialization from `libnfc` ??
- **suspend, powercycle, resume** : 
In this case I'm sure, libnfc and application has to reinitialize the PN7150 just like in normal poweron boot. I will come to this point, when the first one is stable.

My suggestion would be, if driver issues a `pn544_disable`, `libnfc` should be returned a specific errno on a pending or following read/write access. Then `libnfc` could do an appropriate reaction.

jgeslin, 2019-11-05:
Indeed, as said, `pn544_disable`/`pn544_enable` sequence resets the PN7150. This requires the library to be reinitialized.
So, perhaps the only way to handle your power sequence, is to manage it at application level via the linux_libnfc-nci API:
- disabling NFC when suspending by use of  `nfcManager_doDeinitialize()`
- re-enabling NFC when resuming by use of `nfcManager_doInitialize()` and following APIs as `nfcManager_registerTagCallback()` and `nfcManager_enableDiscovery()`.