NXPNFCLinux/nxp-pn5xx #24: Spurious interrupt while using nfcDemoApp

alassaneg, 2022-05-03:
Hello,

I have a eval kit from microchip (sama5d27 ek)  using kernel 5.4 and i'm using the nfcDemoApp demo application to share a text to a NFC reader ( my android phone) and the chip is the PN7150 (NFC 2 click board from mikroe).
When i execute the program, i have a lot of "spurious interupt" display in the terminal while my phone is not even close to the antenna.
When i approach my phone to the antenna, i have even more spurious interrupt and to receive the text from nfcDemoApp, i need to be very patient and move my phone back and forth.
So there is clearly an issue and i don't see where it's coming from. 
I checked with an oscilloscope and actually, PN7150 is generating many many interrupts while my phone is close to the antenna which seems to be a weird behavior.
I'd appreciate any help with that. I have the same issue on a custom board i made.

jgeslin, 2022-05-10:
Hi,
If you look to the nxp-pn5xx implemenation you can observe that the log "spurious interrupt" relates to fake interrupts: interrupt is triggered while related GPIO is not seen HIGH. This is obviously overloading the CPU and impact overall performance of NFC scenario.
It looks the PN7150 IRQ line connected to the MCU is pertuabated by external signals (like phones when it is approached, but not only). I think it is worth to try adding a pull-down resistor, either externally or within the MCU by GPIO pin configuration.

alassaneg, 2022-05-10:
Hi,

Thank you for the answer.
Looking at the scope, GPIO goes high, i can clearly see it. Maybe the GPIO toggle so fast between HIGH and LOW that the driver can't read the GPIO state as HIGH?
I will still try to activate a pull down on this pin.

jgeslin, 2022-05-10:
Hi,

The PN7150 is only driving the IRQ pin back to LOW state when nothing else is to be read. 
If the IRQ line goes HIGH then LOW without any I2C read operation in between, it probably means this is not driven by PN7150 but rather noise or pin conflict (other signal routed on this pin).
The pull-down should resolve first assumption, for second one a shcematic review as well as device-tree analysis should help to figure out.

alassaneg, 2022-05-10:
Actually spurious interrupts happen only when i try to get something being read or send by the PN7150.
I'll still try to activate the pull down pin.

alassaneg, 2022-05-17:
Hello,

I've activated the pull down resistor but i'm still having the spurious interrupt issue.
This is the actual device tree implementation:

` pn547: pn547@28 {
						compatible = "nxp,pn547";
						reg = <0x28>;
						clock-frequency = <400000>;
						interrupt-gpios = <&pioA 32 32>; //PB00
						enable-gpios = <&pioA 34 0>; //PB02
					};
`

Don't really know where to go from here...

jgeslin, 2022-05-18:
Hi,
Can it be that the same interrupt vector is being also associated to another signal ?

alassaneg, 2022-05-22:
No it can't i already checked and it's not that.
Actually, the only thing that causes spurious interrupt is my phone being approched to the NFC antenna to get datas from it.

`  i2c3: i2c@600 {
                                        dmas = <0>, <0>;
                                        interrupts = <20 IRQ_TYPE_LEVEL_HIGH 7>;
                                        i2c-analog-filter;
                                        i2c-digital-filter;
                                        i2c-digital-filter-width-ns = <35>;
                                        pinctrl-names = "default";
                                        pinctrl-0 = <&pinctrl_mikrobus_i2c>;
                                        status = "okay";

                                        pn547: pn547@28 {
                                                compatible = "nxp,pn547";
                                                reg = <0x28>;
                                                clock-frequency = <400000>;
                                                interrupt-gpios = <&pioA 32 GPIO_PULL_DOWN>; //PB0
                                                enable-gpios = <&pioA 34 0>; //PB2
                                        };
                                }; `

alassaneg, 2022-05-25:
When reading these two documents, it's written that, regarding pn5xx driver that "- The detection of spurious interrupts is a known issue, which will be resolved in the next release".
https://www.nxp.com/docs/en/release-note/LS1012A_ASK_1-1-RN.pdf
https://www.nxp.com/docs/en/application-note/AN12354.pdf

Would these spurious interrupt prevent the tag emulation to work well?

jgeslin, 2022-05-25:
Those documents refers to pn5xx porting to LS1012A Layerscape Processor. It relates to issues encountered on this specific processor because of the handling of the IRQ GPIO pin.
This adds arguments pointing to the microchip implementation.
FYI, spurious interrupts should not prevent NFC operations. Only if there are too frequent it can impact the overall transaction time.

alassaneg, 2022-05-25:
I can live with these spurious interrupt i guess while reading tags but whn using using nfcDemoApp it's a nightmare, i'm having a hard time reading NDEF message with my phone since weeks and i'm completely stuck since weeks.
I've tried diffferent phones and still having a hard time.
Do you know what could be the cause?

alassaneg, 2022-05-29:
I've activated debug and this is what happened when using "nfcDemo app share", and here, i wasn't able to receive the NDEF message on my phone.
I have seen various topic on nfc emulation issue but any answer and solution.
Do you see anything strange here?

```
2012:01:01-00:08:23.252 NxpFunc:    nativeNdef_createUri: exit
Waiting for a Tag/Device...

2012:01:01-00:08:25.592 NxpTml:     PN54X - I2C Read successful.....
2012:01:01-00:08:25.592 NxpNciR:     len =  28 > 61051901020400FF010904000408C97C5E0120000000050478807802
2012:01:01-00:08:25.592 NxpTml:     PN54X - Posting read message.....
2012:01:01-00:08:25.593 NxpHal:     read successful status = 0x0
2012:01:01-00:08:25.593 NxpHal:     NxpNci: RF Interface = ISO-DEP
pn54x_dev_read: spurious interrupt detected
2012:01:01-00:08:25.593 NxpHal:     NxpNci: Protocol = ISO-DEP
2012:01:01-00:08:25.594 NxpHal:     NxpNci: Mode = A Passive Poll
2012:01:01-00:08:25.594 NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=28
2012:01:01-00:08:25.594 NxpFunc:    Nfc: Process Event mt:3
2012:01:01-00:08:25.595 NxpFunc:    Nfc: NFC received ntf gid:1
2012:01:01-00:08:25.595 NxpFunc:    Nfc: P2P_Prio_Logic
2012:01:01-00:08:25.595 NxpFunc:    Nfc: nfa_dm_p2p_prio_logic type = 0x2
2012:01:01-00:08:25.595 NxpFunc:    Nfc: ISO-DEP Detected First Time  Resume the Polling Loop
2012:01:01-00:08:25.596 NxpFunc:    Nfc: nfc_ncif_send_cmd.
2012:01:01-00:08:25.596 NxpFunc:    NfcAdaptation::HalWrite
2012:01:01-00:08:25.596 NxpTml:     PN54X - Read requested.....
2012:01:01-00:08:25.597 NxpTml:     PN54X - Invoking I2C Read.....
2012:01:01-00:08:25pn54x_dev_read: spurious interrupt detected
.629    NxpTml:     PN54X - Write requested.....
2012:01:01-00:08:25.629 NxpTml:     PN54X - Invoking I2C Write.....
2012:01:01-00:08:25.632 NxpNciX:     len =   4 > 21060103
2012:01:01-00:08:25.632 NxpTml:     PN54X - I2C Write successful.....
2012:01:01-00:08:25.632 NxpTml:     PN54X - Posting Fresh Write message.....
2012:01:01-00:08:25.632 NxpTml:     PN54X - Tml Writer Thread Running................
2012:01:01-00:08:25.633 NxpHal:     write successful status = 0x0
2012:01:01-00:08:25.634 NxpFunc:    Nfc: Dont process the Event
2012:01:01-00:08:25.636 NxpTml:     PN54X - I2C Read successful.....
2012:01:01-00:08:25.636 NxpNciR:     len =   4 > 41060100
2012:01:01-00:08:25.636 NxpTml:     PN54X - Posting read message.....
2012:01:01-00:08:25.6pn54x_dev_read: spurious interrupt detected
37      NxpHal:     read successful status = 0x0
2012:01:01-00:08:25.637 NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=4
2012:01:01-00:08:25.637 NxpFunc:    Nfc: Process Event mt:2
2012:01:01-00:08:25.638 NxpFunc:    Nfc: NFC received rsp gid:1
2012:01:01-00:08:25.638 NxpFunc:    Nfc: P2P_Prio_Logic
2012:01:01-00:08:25.638 NxpFunc:    Nfc: nfa_dm_p2p_prio_logic type = 0xff
2012:01:01-00:08:25.639 NxpFunc:    Nfc: NFA_DM_RF_DEACTIVATE_RSP
2012:01:01-00:08:25.639 NxpFunc:    Nfc: Dont process the Response
2012:01:01-00:08:25.639 NxpTml:     PN54X - Read requested.....
2012:01:01-00:08:25.639 NxpTml:     PN54X - Invoking I2C Read.....
2012:01:01-00:08:25.709 NxpTml:     PN54X - I2C Read successful.....
2012:01:01-00:08:25.709 NxpNciR:     len =   5 > 6106020300
2012:01:01-00:08:25.710 NxpTml:     PN54X - Posting read message.....
2012:01:01-00:08:25.710 NxpHal:     read successful status = 0x0
2012:01:01-00:08:25.710 NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=5
2012:01:01-00:08:25.711 NxpFunc:    Nfc: Process Event mt:3
2012:01:01-00:08:25.711 NxpFunc:    Nfc: NFC received ntf gid:1
2012:01:01-00:08:25.711 NxpFunc:    Nfc: P2P_Prio_Logic
2012:01:01-00:08:25.711 NxpFunc:    Nfc: nfa_dm_p2p_prio_logic type = 0xff
2012:01:01-00:08:25.712 NxpFunc:    Nfc: NFA_DM_RF_DEACTIVATE_NTF
2012:01:01-00:08:25.712 NxpFunc:    Nfc: P2P_Start_Timer
2012:01:01-00:08:25.712 NxpFunc:    Nfc: Dont process the Event
2012:01:01-00:08:25.712 NxpTml:     PN54X - Read requested.....
2012:01:01-00:08:25.713 NxpTml:     PN54X - Invoking I2C Read.....
2012:01:01-00:08:25.929 NxpFunc:    Nfc: P2P_Timer_timeout NFC-DEP Not Discovered!!
2012:01:01-00:08:25.930 NxpFunc:    Nfc: Deactivate and Restart RF discovery
2012:01:01-00:08:25.930 NxpFunc:    Nfc: nfc_ncif_send_cmd.
2012:01:01-00:08:25.930 NxpFunc:    NfcAdaptation::HalWrite
2012:01:01-00:08:25.931 NxpTml:     PN54X - Write requested.....
2012:01:01-00:08:25.931 NxpTml:     PN54X - Invoking I2C Write.....
2012:01:01-00:08:25.933 NxpNciX:     len =   4 > 21060100
[... 161 lines cut ...]
2012:01:01-00:08:40.599 NxpFunc:    Nfc: Process Event mt:3
2012:01:01-00:08:40.599 NxpFunc:    Nfc: NFC received ntf gid:0
2012:01:01-00:08:40.599 NxpFunc:    Nfc: nci_proc_core_ntf opcode:0x7
2012:01:01-00:08:40.599 NxpFunc:    Nfc: nfa_dm_nfc_response_cback () event=0x500d
2012:01:01-00:08:40.600 NxpFunc:    Nfc: nfa_dm_disc_data_cback ()
2012:01:01-00:08:40.600 NxpFunc:    Nfc: nfa_dm_disc_sm_execute(): state: 1, event:10 disc_flags: 0x1
2012:01:01-00:08:40.600 NxpFunc:    Nfc: nfa_dm_disc_sm_execute(): new state: 1,  disc_flags: 0x1
2012:01:01-00:08:40.600 NxpTml:     PN54X - Read requested.....
2012:01:01-00:08:40.601 NxpTml:     PN54X - Invoking I2C Read.....
2012:01:01-00:08:53.081 NxpTml:     PN54X - I2C Reapn54xcessful.....
2012:01:01-00:08:53.082 NxpNciR:     len =   4 > 600701A1
2012:01:01-00:08:53.082 NxpTml:     PN54X - Posting read message.....
2012:01:01-00:08:53.082 NxpHal:     read successful status = 0x0
[... cut ...]
2012:01:01-00:08:53.084 NxpFunc:    Nfc: nci_proc_core_ntf opcode:0x7
2012:01:01-00:08:53.084 NxpFunc:    Nfc: nfa_dm_nfc_response_cback () event=0x500d
2012:01:01-00:08:53.084 NxpFunc:    Nfc: nfa_dm_disc_data_cback ()
2012:01:01-00:08:53.084 NxpFunc:    Nfc: nfa_dm_disc_sm_execute(): state: 1, event:10 disc_flags: 0x1
2012:01:01-00:08:53.085 NxpFunc:    Nfc: nfa_dm_disc_sm_execute(): new state: 1,  disc_flags: 0x1
2012:01:01-00:08:53.085 NxpTml:     PN54X - Read requested.....
2012:01:01-00:08:53.085 NxpTml:     PN54X - Invoking I2C Read.....
```

jgeslin, 2022-05-30:
From the logs I can see that your device is detecting the phone in card mode (instead of the way around).
Indeed, in the "share" demoApp mode the device is also capable of detecting card.
You can change the behavior by setting POLLING_TECH_MASK to 0x00 inside libnfc-nci.conf file, this will disable remote card detection.

alassaneg, 2022-05-30:
Hi @jgeslin 

Thank you so much for your reactivity.
I just set POLLING_TECH_MASK to 0x00 and it seems to finally work without any issues (despite the spurious interrupt). By the way, log level should be set to 0x00 to make this work.
Thanks again for resolving this, that was not an easy one for me.